genodelabs/genode #2959: dde_linux: remove unused WPA code in src archive

cnuke, 2018-08-31:


nfeske, 2018-09-03:
*Warning about potential data loss!*

The cleanup-wpa rule of the dde_linux.port description used to delete the .git directory to reduce the size of the 3rd-party code. However, this may cause an accidental loss of data when preparing dde_linux twice. When preparing it for the first time, everything is fine.

When preparing it for the second time, the port tool skips the fresh git clone because the directory already exists. Following the clone step, the tool reset the working tree to the git revision specified in the port file. This is where things to wrong. Since the removal of the .git subdirectory at the end of the first prepare, the directory is no longer a valid git repository. This prompts git to walk up the directory tree and apply the given operation to the next git repository found. In our case, this would be the Genode git repository. Git now tries to reset the tree to a commit ID that is no valid in Genode's git repository. So it fails to do that of course. But unfortunately when git detects that, all local changes in the Genode tree are already gone.

cnuke, 2018-09-03:
@nfeske I already addressed this problem differently, please see 9452fd1. Although I prefer my fix, as it deals with the size of src archive rather than the contrib directory, I'm fine with either solution.

nfeske, 2018-09-03:
@cnuke I'm fine with your variant.

My patch was an emergency reaction to data loss experienced by @skalk today. Since the problem may affect any user of the framework, a quick master update was in order.

Let's rebase your commit for the inclusion to staging.

nfeske, 2018-09-03:
@cnuke I added you commit to staging.

cnuke, 2018-09-03:
Not sure if it is worth it but 8e6cf35 introduces a check for _complete_ git checkout.

ehmry, 2018-09-04:
I have a patch to require a safety .git in CONTRIB_DIR at 7fb6f0fb3f37106d9d65e8ad94de14938beab9e1.

chelmuth, 2018-09-04:
Commit 44a11aac0b477773f9f2c8bf24771e77cfb90e83 limits the git tool to the local directory on reset/fetch in the port tool. The question is now, which option we should go for resp. which to combine.

nfeske, 2018-09-04:
Thanks everyone for investigating ways to safeguard us in the future. I'd go for a combination of @chelmuth's (similar in spirit to @cnuke's suggestion but less "shellish") and @ehmry's safety measures.

commit 7dc7c2ca (2018-09-03): dde_linux: remove unused WPA code in src archive

Rather than removing the source files during the port preration, which
breaks preparing the port anew, they are now removed while creating
the src archive.

Fixes #2959.
commit b93f1504 (2018-09-05): dde_linux: remove unused WPA code in src archive

Rather than removing the source files during the port preration, which
breaks preparing the port anew, they are now removed while creating
the src archive.

Fixes #2959.
closed 2018-09-05 by nfeske