genodelabs/genode #2962: porting 3rd-software problem

jenkin233, 2018-09-04:
Hello,
I am new to genode and  I am porting a 3rd-software which is a standard POSIX/Libc application to genode. I followed this tutorial: [https://genode.org/documentation/developer-resources/porting_dosbox](url) and add "LIBS+ = posix" in my target.mk file. When I want to test it, the compilation and link process seems successful , but when it run with qemu, error occurred like this:

`[init` -> mytest] Error: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="posix.lib.so")
[init` -> mytest] Error: Could not open ROM session for "posix.lib.so"
[init -> mytest] Error: LD: exception during program load: 'Genode::Rom_connection::Rom_connection_failed'
[init -> mytest] Error: Uncaught exception of type 'Genode::Rom_connection::Rom_connection_failed' `
I guess it's because something wrong with my run script, which looks like this:
`set build_components {
  core init
  app/mytest
}
build $build_components

create_boot_directory

install_config {
 <config>
   <parent-provides>
     <service name="ROM"/>
     <service name="RAM"/>
     <service name="CAP"/>
     <service name="PD"/>
     <service name="RM"/>
     <service name="CPU"/>
     <service name="LOG"/>
   </parent-provides>
   <default-route>
     <any-service> <parent/> <any-child/> </any-service>
   </default-route>
   <start name="mytest" caps="200">
     <resource name="RAM" quantum="128M"/>
     <config>
       <libc stdout="/dev/log" stderr="/dev/log">
       </libc>
     </config>
   </start>
 </config>
 }

 set boot_modules {
   core init ld.lib.so vfs.lib.so
   libc.lib.so libm.lib.so
   pthread.lib.so darknet
 }
 build_boot_image $boot_modules

 append qemu_args " -m 256"
 run_genode_until forever`
Actually , I am not sure how to write a run script to  make this  application run, and there's quite more question I want to know (for example, the application's `main` function takes several parameters to run, however, how this parameters can be passed to the porting application in genode?). I looked up the document about genode ( genode-foundation-18.05.pdf), but find little related to this. Can anyone give me some help?
Thanks.

chelmuth, 2018-09-04:
Please make sure that *posix.lib.so* is in your boot modules, e.g.,
```tcl
build_boot_image { core init timer ... ld.lib.so libc.lib.so vfs.lib.so libm.lib.so posix.lib.so }
```

chelmuth, 2018-09-04:
Information about how to provide POSIX environment variables or command-line arguments can be gathered in *repos/libports/run/libc_getenv.run*. As there's unfortunately no comprehensive documentation, please also refer to the following example. Executing

```sh
export foo=bar; test-libc_getenv foo
```

on Linux is similar to the following init config.

```xml
<start name="test-libc_getenv">
	<resource name="RAM" quantum="4M"/>
	<config>
		<vfs> <dir name="dev"> <log/> </dir> </vfs>
		<libc stdout="/dev/log"/>
		<env key="foo" value="bar"/>
		<arg value="test-libc_getenv"/>
		<arg value="foo"/>
	</config>
</start>
```

jenkin233, 2018-09-04:
Thanks for your quick reply, that's works for me !
And I have a few more questions expect to get answers:
1. How can I get  input  in the application.
In my application ,there's `scanf` function, but in my test, it happens nothing, just like skipped it. I know for  `printf` function, I can add `stdout='/dev/log` for my output, but what about input? Should I add something like `/dev/xxx` or should I use an input driver?(It seems input driver respond to every key press/release?)
2. How can application read the file?
In my application, I want to read a file 'file.txt', I add `<inline name="file.txt"> </inline>` in the config and add `file.txt` to boot modules in my run script, but it seems  it virtual out a file called 'file.txt' and have nothing to do with my own file.
Can you give me some help? Hope I didn't bother you too much.

nfeske, 2018-09-21:
@jenkin233 to get input into an application, you need at least an input driver and some kind of terminal. With a terminal in place you may mount a terminal session to the VFS (e.g., at /dev/terminal) of your program and let the libc use this terminal session via `stdin="/dev/terminal"`.

For reading the content of a ROM module as a file, you have to add a `<rom>` node to the VFS, not an `<inline>` node (this one makes the node content available as file content).

BTW, questions like these are best directed to our mailing list (https://genode.org/community/mailing-lists). The issue tracker is mostly used for reporting bugs or requesting improvements.

Would you mind closing the issue and continue the discussion on the mailing list?

jenkin233, 2018-09-21:
OK, thanks for your reply. I 'll close it.

closed 2018-09-21 by jenkin233