genodelabs/genode #2963: sculpt: usb mouse non functional when running in Virtualbox

alex-ab, 2018-09-05:


alex-ab, 2018-09-05:
Beside fixing the regression in the USB driver by commit 6baf9e3371f4825ce21aa3d1f3c53d19e9b85456, the mouse will still not work properly in Virtualbox. The reason is that Virtualbox provides a USB OHCI absolute pointing device, which produces x/y coordinates up to 32k. The screen is typically smaller, which means we would need a transformation. In principle the usb driver supports this already, however a more generalized transformation mechanism is wished (e.g. in/by the input_filter). For now, the driver_manager may also switch off USB OHCI when running in Virtualbox - see c1cc07cba2219c5bcbb78d41cfb47d9f0d5404a9.

nfeske, 2018-09-05:
@alex-ab The driver-manager patch looks very good in my opinion. Thanks for investigating this issue so quickly!

skalk, 2018-09-05:
Well, does that mean we need to extend the new usb_host_drv  to be able to enable/disable different host controller models too, or is there a short-term plan to build a generic solution (input-filter)?

nfeske, 2018-09-05:
There is no short-term plan to address the calibration of touch/tablet events. @skalk it might be sensible to add an 'ohci="no"' option to the USB host driver.

commit f01464ef (2018-09-05): driver_manager: disable ohci when inside vbox

Fixes #2963