genodelabs/genode #2966: core: fatal errors when building with -O0

chelmuth, 2018-09-05:
According to my tests O0 builds produce the following errors on x86_64 during `make run/libc`

**KERNEL=nova**
```
Error: Uncaught exception of type 'Genode::Reconstructible<Genode::Account<Genode::Ram_quota> >::Deref_unconstructed_object'
Warning: abort called - thread: entrypoint
Error: 
PAGE-FAULT IN CORE addr=0x0 ip=0x268f01 (read)

stack pointer 0xa01fd500, qualifiers 0x4 irUwp
faulter utcb 0xa01ff000, last message item count 0
  #1 0x00000000a01fd500 0x0000000000268f01
  #2 0x00000000a01fd528 0x0000000000116a20
```

**KERNEL=hw**
```
Error: Uncaught exception of type 'Genode::Reconstructible<Genode::Account<Genode::Ram_quota> >::Deref_unconstructed_object'
Warning: abort called - thread: entrypoint
Error: core -> entrypoint raised a fault, which should never happen ip=0xffffffc000165121 fault-addr=0x0 type=no-page
```

More simple test scenarios (e.g., log or rm_nested) work. I did not test other kernels or architectures.

chelmuth, 2018-09-05:
**x86_32 KERNEL=nova**
```
Error: Uncaught exception of type 'Genode::Reconstructible<Genode::Account<Genode::Ram_quota> >::Deref_unconstructed_object'
Warning: abort called - thread: entrypoint
Error: 
PAGE-FAULT IN CORE addr=0x0 ip=0x1dfc18 (read)

stack pointer 0xa01fddb0, qualifiers 0x4 irUwp
faulter utcb 0xa01ff000, last message item count 0
  #1 0xa01fddb0 0x001dfc18
  #2 0xa01fddb4 0x001e9c28
```

**x86_32 KERNEL=okl4**
```
Error: Uncaught exception of type 'Genode::Reconstructible<Genode::Account<Genode::Ram_quota> >::Deref_unconstructed_object'
Warning: abort called - thread: entrypoint
roottask read pagefault at 00000000, ip=030aeaa9 - deny
--- "KD# roottask pagefault" ---
--------------------------------- (eip=f0111fc1, esp=f011de44) ---
```

I'd resume that the bug is portable.

alex-ab, 2018-09-06:
e41d6efdb632bf37c2158069ae003189350496d4 fixes it for me

commit a169bfeb (2018-09-07): core: destruct pd::_ram_quota after regions

Fixes #2966
commit 2b2d7507 (2018-09-13): core: destruct pd::_ram_quota after regions

Fixes #2966