genodelabs/genode #2967: Xml_generator: fix exception handling in Node(...)

m-stein, 2018-09-07:


nfeske, 2018-09-07:
This issue title, description, and label leaves me puzzled. It suggests that something fundamental is broken and must be fixed but remains completely unspecific about the symptom and nature of the problem.

If the issue is about the topic we discussed offline just yesterday, an appropriate title would be "Xml_generator: improve formatting on exception" with the label "cleanup" attached. From what I know, the issue is merely about a formatting improvement in the exception case, which happened to become a relevant in your case where exceptions are used in a non-exceptional way.

I want to encourage you to spend a little more effort on meaningful issue titles, descriptions, and labels. An issue should at least be specific about the symptom and the expected behavior. Please keep in mind that the issue tracker is part of our project's documentation.

m-stein, 2018-09-07:
@nfeske Sorry for that. I thought it would be fixed right away but it turned out to be more time-intensive.

nfeske, 2018-09-07:
The commits are very insightful. Thanks for the good commit messages and the extended tests.

chelmuth, 2018-09-11:
I suspect the commit series breaks `make run/report_rom`, please check.

nfeske, 2018-09-11:
The commit `Xml_generator: fix and test missing '\0'` causes a segfault in init in the leitzentrale.run scenario on base-linux.

nfeske, 2018-09-11:
Let's keep this line of work for merging at a later point because it deserves more attention than I'm able to provide right now.

@m-stein I'm wondering, in the commit message you state "Without exceptions in the Node constructor this led to "\0\n" at the end of generated XML." as a fact. But have you actually witnessed this sequence in the wild? I'm asking because I haven't (looking at the reports in sculpt's report fs), and when looking at the code, I would not expect it either.

Let my try to clarify, the `_out_buffer.append('\')` operation is applied to a local copy of the `_out_buffer` _after_ committing the content. This local copy (the buffer meta data, not the characters) is discarded afterwards. Hence, the zero-character is inserted but it stays outside the committed character sequence.

BTW, The `\n` at the end ensures that the generated data ends with a newline when printed, e.g., by `cat /report/runtime/state`. I would like to preserve it. (I noticed that you added a manual `\n` to the test, which I find odd). But it actually overwrites the last written `\0`. So it might be sensible to also append a `\0` here. That said, we normally don't depend on the zero-termination of the generated string but rather propagate the number of generated characters to the user via the `user()` accessor (see the use in repos/os/include/os/reporter.h). So I'm still puzzled how your patch could introduce the segfault problem I reported for `leitzentrale.run`.

...to be investigated later.

nfeske, 2019-01-30:
Fixed in master.

commit 9dc4c9d9 (2018-09-10): Xml_generator: fix and test missing '\0'

Previously, a '\0' was appended to the out buffer at the end of each
call of the Node constructor. At the end of the Xml_generator
constructor, at the other hand, '\n' was appended. Without exceptions in
the Node constructor this led to "\0\n" at the end of generated XML.
However, with exceptions, the '\0' of the last Node constructor call was
sometimes overwritten, leading to a mere "\n" at the end. It is better
to append '\0' only in one place, namely at the point where the '\n' was
appended in the Xml_generator constructor. This also fixes the probl [... cut]
closed 2019-01-30 by nfeske