genodelabs/genode #2971: Libc: large sends fail

ehmry, 2018-09-10:
The libc fails to make large sends to sockets, particularily if the send or write is larger than the size of an intermediate File_system buffer or the TCP stream send window.

ehmry, 2018-09-11:
Fixed on my `issue2971_buffered_write` branch, but requires the VFS error cleanup of #2954.

ehmry, 2018-09-11:
Still not fixed for `O_NONBLOCK` writes, I need to do some research there.

Actually it seems that POSIX technically allows us to return short writes everywhere.

ehmry, 2018-09-11:
I think that the VFS should perform short writes when there is insufficient buffer space to complete the write in a single operation. This would be consistent with POSIX:

> If a write() requests that more bytes be written than there is room for (for example, the file size limit of the process or the physical end of a medium), only as many bytes as there is room for shall be written. For example, suppose there is space for 20 bytes more in a file before reaching a limit. A write of 512 bytes will return 20. The next write of a non-zero number of bytes would give a failure return (except as noted below).

Or atleast we should do this for non-blocking write descriptors:

> When attempting to write to a file descriptor (other than a pipe or FIFO) that supports non-blocking writes and cannot accept the data immediately:
>
> - If the O_NONBLOCK flag is clear, write() shall block the calling thread until the data can be accepted.
>  - If the O_NONBLOCK flag is set, write() shall not block the thread. If some data can be written without blocking the thread, write() shall write what it can and return the number of bytes written. Otherwise, it shall return -1 and set errno to [EAGAIN].

http://pubs.opengroup.org/onlinepubs/9699919799/

The problem here is that the writes no longer need to return immediately, which is what we have been avoiding.

chelmuth, 2018-09-12:
Nevertheless, we hav to maintain BSD socket semantics in libc via socket fs too, which are for send()

> The length of the message to be sent is specified by the length argument. If the message is too long to pass through the underlying protocol, send() shall fail and no data shall be transmitted.
> 
> Successful completion of a call to send() does not guarantee delivery of the message. A return value of -1 indicates only locally-detected errors.

ehmry, 2018-09-14:
The libc_vfs test checks larges writes with the 5ff3ff57d5134b04e006f85a0d136442e33b9087 patch, and the libc write loop is implemented in d67d4b94b9673d03a8442c23f54119a7256ca33c.

ehmry, 2018-09-17:
I have fixups on my `vfs` branch for the netperf tests. The changes to libc are to break write loops when a plugin returns a zero write but not `WRITE_BLOCKED`, the changes to lwip are to notify the application on TCP ACKs to unblock writes waiting on a full send buffer, and the changes to lxip are to return zero for any `sendmsg` error from the Linux stack.

chelmuth, 2018-09-17:
Please feel free to merge your fixups into the original commits as I postponed the integration into staging of until after Sculpt VC (presumably end of this week).

ehmry, 2018-09-18:
Methinks returning short writes to the application, even those under 16K, should happen. Send will be a special I/O case for anyway. If we introduce signals into the libc, we will have to come back to this topic to deal with *EINTR*.

ehmry, 2018-09-21:
In any case, the Lwip plugins needs to block when writing. Every other plugin does this.

alex-ab, 2019-07-10:
With commit 0f9fa5bf8499267bb99a9932c0b69cd18713fe8e on staging tonight 15 of 17 vbox5 tests failed. After reverting the commit the one tested Linux VM worked again.
There are not obvious output from the Genode side, but the guest complains:
```
error: attempt to read or write outside of disk 'hd0'
Entering rescue mode...
grub rescure>
```

chelmuth, 2019-07-10:
Thanks, I reverted the commit with 269389cab78af598dbddcbc57203444a346b9a3d until a fix is available.

ehmry, 2019-07-11:
I'm not able to reproduce an error like this with GRUB or Syslinux guests on Sculpt, perhaps this is due to a difference in behavior between rump_fs and vfs/rump?

FWIW one of the errors is near a comment refering to VirtualBox: http://git.savannah.gnu.org/cgit/grub.git/tree/grub-core/disk/i386/pc/biosdisk.c#n459

chelmuth, 2019-07-15:
Sorry, but again all vbox tests failed with 26a64929319c1e4d2cfda7252ddd21e9a7b05d83, c752be0176e53d4fdf57a0601d5ef61c996ceee3, and 0ea825b8b30a1e65df933f56b697fb9ee0a4cff8 applied.

chelmuth, 2019-08-13:
I moved all three commits from staging to a topic branch https://github.com/chelmuth/genode/commits/issue_vfs_pipe_rump_fs as we still have issues with our virtualbox tests and currently no resources to investigate until release 19.08. This puts this issue as well #3447 and #2303 on hold.

nfeske, 2019-11-19:
Fixed by fab2fc874f178f7b354efb9ea57fd32941745cab at master.

commit 0f9fa5bf (2019-07-09): Add write completion loop to VFS server and libc

Loop at the VFS server and libc VFS plugin on partial writes. This
implies that the VFS server will not process successive packets for an
open node until a write packet has been processed completely. The libc
will now supspend and reissue write requests to the VFS until the write
has been submitted completely.

Ref #2303
Fix #2971
commit c752be01 (2019-07-12): Add write completion loop to VFS server and libc

Loop at the VFS server and libc VFS plugin on partial writes. This
implies that the VFS server will not process successive packets for an
open node until a write packet has been processed completely. The libc
will now supspend and reissue write requests to the VFS until the write
has been submitted completely.

Ref #2303
Fix #2971
closed 2019-11-19 by nfeske