genodelabs/genode #2973: wifi_drv: regression in Intel 6xxx cards

NobodyIII, 2018-09-10:
As stated in the 18.08 release notes, there has been a regression in wifi_drv's support for Intel 6xxx cards. I have an Intel 6300 card in my laptop, and it no longer works properly with wifi_drv. In particular, when connecting, it never gets an IP address. It authenticates properly and claims to connect, but it fails to configure IPv4. (Genode's drivers seem to lack support for IPv6, so I can't test that.)

According to Josef Söntgen on August 31:
"It looks like some kind of race or timing issue as most of the time the
packets submitted to the firmware are not acknowledged but sometimes it
works just fine and the behaviour is influenced by debug messages."

m-stein, 2018-09-12:
@chelmuth These should fix the NIC router uplinks test on x86:
209276fac75954912200c6a8c1481c63478e32b8 wifi_drv: fix tasklet_schedule
5c25006 nic_router_uplinks.run: make it easier to debug
cc3b606 nic_router_uplinks.run: adapt to new wifi driver

NobodyIII, 2018-09-12:
The commit works. Thanks!

commit 209276fa (2018-09-12): wifi_drv: fix tasklet_schedule

The tasklet_schedule implementation was missing the unblock call, which
was no problem in the past because the correspending task would get un-
blocked by other work items. Now that every workqueue has its own task,
there is nobody left to unblock the task and thereby execute the
tasklet. As a tasklet is the only way to transmit any pending frames
after transmission was temporarily suspended - because the device/fw
was not able to process the TX request - this rendered 6xxx devices
unusable whenever this situation occurred.

Fixes #2973.