genodelabs/genode #2979: Depot_deploy will not route abritrary session types

ehmry, 2018-09-13:
Deploy_deploy is not routing arbitrary sessions, that is, sessions not defined in the canonical Genode framework.

nfeske, 2018-09-13:
This somehow breaks the abstraction that I deliberately introduced in the runtime definitions. By making the supported session types an explicit list, I want to achieve the following:
* Avoid the proliferation of new (use-case-driven and most certainly ad-hoc designed) session interfaces. Requirements that cannot be fulfilled by the existing session interfaces should be used to refine the existing interfaces, or taken as the basis for a new generic and official interface.
* Concise XML notation (instead of `<service name="Nitpicker"/>`, its just `<nitpicker/>`)
* Well defined semantics (defined set and meanings of the various session interfaces) of runtime definitions that can be tested against an XML schema. This anticipates the non-human processing and generation of runtime definitions.

ehmry, 2018-09-13:
I suppose I can run a private version of depot deploy, my session interface is not binary stable and is for the most part a private project.

closed 2018-09-13 by ehmry