genodelabs/genode #2984: sculpt: add Arora runtime packages and launcher

cproc, 2018-09-13:
Some extra notes about the current state:

- the Arora package from my depot (referenced by the launcher) uses a git commit for non-blocking 'connect()' [1] which is still work in progress
- when using the browser in parallel with a VirtualBox VM, the VM can become temporarily unresponsive because of the lower priority of some VirtualBox threads
- the search text input field right next to the address input field doesn't seem to work

[1] https://github.com/cproc/genode/commit/75b903f35a76dc6d63a0f9bd148476dbfc82467b

nfeske, 2018-09-14:
Thank you for the convenient package!

I would not present the ancient demo page by default though. May you tweak the package to show a blank page instead? Or point it at genode.org?

nfeske, 2018-09-14:
While browsing with Arora, I encounted two effects:

* After opening a fairly complex website (sz-online.de), my VM stumbled with the following assertion:
  ```
  EMT-3    !!Assertion Failed!!
  EMT-3    Expression: u64Now <= u64VirtualNow - offSyncGivenUp
  EMT-3    Location  : /pla
  [runtime -> vm -> vbox] in/no/genode.git/depot/nfeske/src/vbox5-nova/2018-09-11/src/app/virtualbox/src/VBox/VMM/VMMR3/TM.cpp(2383) void tmR3TimerQueueR
  [runtime -> vm -> vbox] unVirtualSync(PVM)
  [runtime -> vm -> vbox] 
  [runtime -> vm -> vbox] !!Assertion Failed!!
  [runtime -> vm -> vbox] Expression: u64Now <= u64VirtualNow - offSyncGivenUp
  [runtime -> vm -> vbox] Location  : /plain/no/genode.git/depot/nfeske/src/vbox5-nova/2018-09-11/src/app/virtualbox/src/VBox/VMM/VMMR3/TM.cpp(2383) void tmR3TimerQueueRunVirtualSync(PVM)
  ```
  This seems to be related to @cproc's remark about the possible starvation of vbox caused by its internal use of priorities. Even though we cannot fix the starvation right now, it would be good to somehow avoid the assertion, which kills the entire VM instead of just making it temporarily unusable.

* When opening old.reddit.com/r/programming, Arora exited with the following error:
  ```
  [runtime] [34mWarning: arora: no route to service "ROM"[0m
  [runtime -> arora -> arora] [31mError: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="egl_drv.lib.so")[0m
  [runtime -> arora -> arora] [31mError: Could not open ROM session for "egl_drv.lib.so"[0m
  [runtime -> arora -> arora] Error: could not open EGL back end driver ('egl_drv.lib.so')
  [runtime -> arora -> arora] libEGL debug: EGL user error 0x3001 (EGL_NOT_INITIALIZED) in eglInitialize
  [runtime -> arora -> arora] 
  [runtime -> arora -> arora] eglInitialize() failed
  [runtime -> arora -> arora] [34mWarning: kill: kill not implemented[0m
  [runtime -> arora -> arora] [34mWarning: kill: kill not implemented[0m
  [runtime -> arora -> arora] [34mWarning: 12 dangling allocations at allocator destruction time[0m
  [runtime -> arora -> arora] [34mWarning: 2 dangling allocations at allocator destruction time[0m
  [runtime -> arora -> arora] [34mWarning: 177 dangling allocations at allocator destruction time[0m
  [runtime -> wm -> wm] [34mWarning: submitting signal failed - error 4 - context=cap<36160>[0m
  [runtime -> arora] child "arora" exited with exit value 1[0m
  ```
  Maybe, we could avoid this failure by shipping a missing library as part of the pkg?

alex-ab, 2018-09-14:
> Even though we cannot fix the starvation right now, it would be good to somehow avoid the assertion, which kills the entire VM instead of just making it temporarily unusable.

I see 2 (immediate) ways, either
- disable assertion in the VBox5 build (and hope that it survives ...)
- put Arora to a prio level below the VBox VMs

nfeske, 2018-09-14:
I find the first option somehow tempting.

Actually, it is quite nice that the native Arora cannot be influenced by the guest OS. This makes the interactive performance quite deterministic. Once the page is loaded, Arora is indeed quite snappy, regardless of what the VM is doing.

cproc, 2018-09-14:
Fixup commit c18294b fixes the egl driver problem and my updated package shows a blank start page.

nfeske, 2018-09-17:
@alex-ab How do you think about tweaking the build description files of vbox at the packaging-step to disable assertions? This way, assertions remain present for run scripts that build vbox directly but not when deployed from the official packages.

alex-ab, 2018-09-17:
f20b4ccc19ccf810828d494310ee0508ff3ff35b disables assertions for the packaging step - without any promise in whatever direction.

closed 2018-09-21 by chelmuth

nfeske, 2018-09-21:
@alex-ab thanks to your patch 142d503cb4b4ebacb4c4d106c227429d471926d9 my guest OS survived the starvation situation where I opened sz-online.de in the natively running Arora. Thank you!

commit af93315c (2018-09-18): sculpt: add Arora runtime packages and launcher

Fixes #2984
commit 654f6e44 (2018-09-20): sculpt: add Arora runtime packages and launcher

Fixes #2984
commit cc92636d (2018-09-21): sculpt: add Arora runtime packages and launcher

Fixes #2984