genodelabs/genode #2985: nova: relax static kernel memory configuration

alex-ab, 2018-09-13:
For the NOVA kernel memory one may define statically during link time the upper amount of kernel memory to be used. Currently Genode configures 32M for the kernel. Depending on the load and the target machine, the choosen value may not fit.

Especially in the event of Sculpt, we neither know the system memory amount of the machines beforehand (512M...32G+++) nor do we know the load the user will put on Sculpt (e.g. running several VMs at the same time as done by several developers).

Instead of choosing static kernel memory values during link time, the kernel may allocate some amount based on the available system memory.

alex-ab, 2018-09-13:
The commit works for me on 3 machines. It would be good if we can get some more test results before merging.

NobodyIII, 2018-09-13:
What should the upper limit really be and why? 1G seems big for a microkernel's reserved memory.

alex-ab, 2018-09-14:
> What should the upper limit really be and why?

There is no satisfying answer to that, if you have a kernel which takes statically a part of the system memory as kernel memory (as NOVA does, unfortunately). You ever may construct a scenario, where you either have too less or too much memory allocated for the kernel.

If you don't like this, you need a dynamic way of allocating kernel memory, as e.g. seL4 or our hw does in principle.

> 1G seems big for a microkernel's reserved memory.

Sure. The numbers are just a heuristic we can adjust.

trimpim, 2018-09-14:
I have (for a short time only) tested this patch on two machines we have here (one 20GB  the other 8GB of memory).
On both I was bale to start multiple virtual machines. So far I could work on the machines in parallel and did not have any problems.

The only thing of which I'm not sure are the following log messages I get on the machine with 20GB of ram (Lenovo T460s):
[kernel] 03ae O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303b2 RB:0x000e03b2 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303b5 RB:0x000e03b5 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303b9 RB:0x000e03b9 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303b9 RB:0x000e03b9 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303bd RB:0x000e03bd O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303c1 RB:0x000e03c1 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303c5 RB:0x000e03c5 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303c9 RB:0x000e03c9 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303cd RB:0x000e03cd O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303d1 RB:0x000e03d1 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303d5 RB:0x000e03d5 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303d9 RB:0x000e03d9 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000303e7 RB:0x000e03e7 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x0003042d RB:0x000e042d O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x00030431 RB:0x000e0431 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x00030435 RB:0x000e0435 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x00030438 RB:0x000e0438 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x0003043d RB:0x000e043d O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x00030444 RB:0x000e0444 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x00030448 RB:0x000e0448 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x00030460 RB:0x000e0460 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x00030464 RB:0x000e0464 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x00030468 RB:0x000e0468 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000301e5 RB:0x000e01e5 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000301e9 RB:0x000e01e9 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000301ed RB:0x000e01ed O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000301f1 RB:0x000e01f1 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000301f4 RB:0x000e01f4 O:0x00 A:0x7 SUB:2
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000301f7 RB:0x000e01f7 O:0x00 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff8890e940->0xffffffff82475280 SB:0x000301fb RB:0x000e01fb O:0x00 A:0x7 SUB:2

I'm not entirely sure but it might be, that they are also present without the patch.

alex-ab, 2018-09-14:
Thanks for testing. The overmap messages we have seen also before the patch already.
For now I have another machine which needs some investigations.

alex-ab, 2018-09-17:
I reworked the commit to foster some more machines. The allocation parameters may now be configured outside the source code to some degree. It would be good if you can re-test the commit on your machines.

trimpim, 2018-09-18:
I tested your latest patch on my machine with 20GB of RAM. I ddn't have time yet to test it on the machine with 8GB.
So far  everything seems to work as it should.

alex-ab, 2018-09-18:
Thanks for testing. From my side the issue is ready.

alex-ab, 2018-09-19:
@chelmuth: added a message about the kernel memory to the log exported by core

alex-ab, 2018-09-19:
@chelmuth: sorry, had to force push due to 32bit compilation issue

chelmuth, 2018-09-19:
@alex-ab no prob, picked 380c9e2f92

trimpim, 2018-09-20:
I have tested some more on my machine with 20GB of RAM.
I have strange performance issues when I start a windows 10 in a virtual machine with the patch applied. Without the patch windows boots without a problem, when I boot with the patch, the boot animation is very slow, and windows doesn't boot.

I can't think of a reason why this should be the case though.

chelmuth, 2018-09-20:
@trimpim Please make sure you're using the latest version of the commit. Do you have similar issues with Linux VMs?

trimpim, 2018-09-20:
@chelmuth I just built it with the latest version of the patch. The boot animation of Windows still is very slow.
Linux also is a bit slower, but I am still able to work in the virtual machine.

I did not change the settings in the target.mk. This means, that on my machine nova will use about 200M of RAM. In my other configuration I have set the value to 128M.

alex-ab, 2018-09-20:
@trimpim: this performance issue I had with a earlier commit, the latest commit should not have it. With the latest patch core tells you the kernel memory in the log (and also on sculpt in report/log) - so you have not to guess the amount of kernel memory. If this line is missing, you won't have the latest commit (or maybe it is not rebuild or you take some older state of pkg ...)

alex-ab, 2018-09-20:
So the git commit in the hypervisor line should be b156131 and the kernel memory entry:
```
NOVA Microhypervisor v8-b156131 (x86_64): Sep 20 2018 20:11:46 [gcc 6.3.0] [MBI2]
...
 ROM:
 ROM: 
...
36M kernel memory

Genode 18.08-63-gb71d094
472 MiB RAM and 63253 caps assigned to init
```

trimpim, 2018-09-21:
@alex-ab I did indded not have the log messages in `report/log`, so I checked if I made an error when integrating your patch, but this was not the case (`repos/base-nova/ports/nova.port`uses the commit `b156131007c928e2968fc5e551cea402c1886919`).
The next thing I suspected was that I made an error while copying the content of `var/run/sculpt/boot/*` to `/boot/` on the USB stick, so I used `dd` to recreate the whole USB stick.
Still I do not have the LOG entries.
So I cleaned all previous build artifacts (including the `depot` and `public`) and do a full rebuild. I'll give a feedback as soon this is finished.

trimpim, 2018-09-21:
Now, after the complete rebuild, windows boots again.
Many thanks for the support.

commit 311250be (2018-09-19): nova: define kernel memory based on system memory

Switch to kernel branch, that determines the available system memory during
boot time. The overall kernel memory is still static, but during boot time
dynamically the amount can be chosen. Following 3 config option exists:

CONFIG_MEMORY_BOOT is the amount of kernel memory allocated in the BSS
statically - effectively chosen during link time - see linker script.

CONFIG_MEMORY_DYN_MIN && CONFIG_MEMORY_DYN_PER_MILL configures the dynamic
part of the kernel memory allocation applied during early kernel boot time.
CONFIG_MEMORY_DYN_MIN is the amo [... cut]
commit 43515296 (2018-09-20): nova: define kernel memory based on system memory

Switch to kernel branch, that determines the available system memory during
boot time. The overall kernel memory is still static, but during boot time
dynamically the amount can be chosen. Following 3 config option exists:

CONFIG_MEMORY_BOOT is the amount of kernel memory allocated in the BSS
statically - effectively chosen during link time - see linker script.

CONFIG_MEMORY_DYN_MIN && CONFIG_MEMORY_DYN_PER_MILL configures the dynamic
part of the kernel memory allocation applied during early kernel boot time.
CONFIG_MEMORY_DYN_MIN is the amo [... cut]
commit 3824ad47 (2018-09-21): nova: define kernel memory based on system memory

Switch to kernel branch, that determines the available system memory during
boot time. The overall kernel memory is still static, but during boot time
dynamically the amount can be chosen. Following 3 config option exists:

CONFIG_MEMORY_BOOT is the amount of kernel memory allocated in the BSS
statically - effectively chosen during link time - see linker script.

CONFIG_MEMORY_DYN_MIN && CONFIG_MEMORY_DYN_PER_MILL configures the dynamic
part of the kernel memory allocation applied during early kernel boot time.
CONFIG_MEMORY_DYN_MIN is the amo [... cut]
closed 2018-09-21 by chelmuth