genodelabs/genode #2988: wifi_drv: add hidden network support

cnuke, 2018-09-17:


cnuke, 2018-09-18:
@chelmuth please merge cce6da to staging.

cnuke, 2018-09-19:
@chelmuth please merge 32e4064 to staging.

chelmuth, 2018-09-19:
I'm afraid the following limit will hit us pretty soon (hit it just by playing around with the network list). https://github.com/genodelabs/genode/blob/c35d7c4fa73a4a7d87e797922abc320496602e22/repos/dde_linux/src/drivers/wifi/frontend.h#L254-L256 Couldn't we go for a dynamic data structure?

cnuke, 2018-09-19:
I would not say soon because the limit will only get reached iff you have more than 16 auto-connect enabled and valid networks in your configuration but sure, it can be changed to use a Registery.

cnuke, 2018-09-19:
Ah right, the semantics changed when enabled became auto-connect so it is probably more likely.

chelmuth, 2018-09-20:
So, I consider this as *yes* and expect a follow up commit before we finalize this topic.

chelmuth, 2018-09-21:
Some commits pending...

commit caaf7dc1 (2018-09-18): wifi_drv: add hidden network support

Fixes #2988.
commit ecccbb46 (2018-09-20): wifi_drv: add hidden network support

Fixes #2988.