genodelabs/genode #2990: vfs_import copies to the root directory when inside a subdirectory

NobodyIII, 2018-09-18:
Currently, the vfs_import plugin always copies to the root directory, regardless of whether it is placed inside the root directory or inside a subdirectory. This is a big limitation that has kept me from using it to test my own VFS plugin, and is also highly counter-intuitive.

NobodyIII, 2018-09-18:
To be clear, I have been speaking of this as a limitation rather than a bug because the code currently makes no attempt to copy outside of the root directory. Currently, there seems to be no clean way for a VFS plugin to detect its own path.

nfeske, 2018-09-18:
Yes, the `<import>` node is only meant to reside within the `<vfs>` node directly. I see how this is a bit unintuitive but - in my opinion - it is not worth spending much effort in changing that.

You can already use the `<import>` plugin to populate any subdirectory in the VFS by using `<dir>` nodes inside the `<import>` node. E.g., the following example imports the ROM "config" into the `/tmp/` directory.

```
<vfs>
  <dir name="tmp"> <ram/> </dir>
  <import>
    <dir name="tmp">
      <rom name="config"/>
    </dir>
  </import>
</vfs>
```

ehmry, 2018-09-18:
The **import** plugin does not know its location in the VFS heirarchy, and it doesn't matter because it has no location, the plugin does not provide any files.

nfeske, 2018-09-21:
May we close this issue?

NobodyIII, 2018-09-21:
Yes. Thanks for the workaround. I'm not sure why I didn't think of that myself.

closed 2018-11-01 by NobodyIII