genodelabs/genode #2993: sel4/arm: alignment faults may lead to endless mapping loops

alex-ab, 2018-09-21:


alex-ab, 2018-09-21:
With the commit one may determine that an repeating alignment fault happens, however it would not stop the endless loop between core and faulting thread. Is this desired and if yes, where should I stop the loop. https://github.com/genodelabs/genode/blob/master/repos/base-sel4/src/core/platform_pd.cc#L151 looks like a place, but for this I would have to propagate the unalignment fault info through several install_mapping methods (platform_thread, platform_pd)

chelmuth, 2018-09-21:
Now that base-sel4 is able to distinguish between present, write, and alignment faults, I suggest to leave the handling of the latter to exception handlers outside of core (like it has to be done for division by 0). So, the first response should be to keep the faulter blocked.

Regarding the actual issue with the USB driver, we have to potential directions to go. The first (and IMO better) option is to enable unaligned accesses on i.MX6 in sel4. Alternatively, we could try to debug the driver and investigate it can be tweaked to use aligned accesses only.

chelmuth, 2018-09-21:
Hm, we should definitely go for the first option as we disable alignment checks in bootstrap for hw on ARM.
https://github.com/genodelabs/genode/blob/5dcf06d2085093d7c079b593c76fc4520d91a3c3/repos/base-hw/src/bootstrap/spec/arm/cpu.h#L25-L35

alex-ab, 2018-09-21:
> Now that base-sel4 is able to distinguish between present, write, and alignment faults, I suggest to leave the handling of the latter to exception handlers outside of core (like it has to be done for division by 0). So, the first response should be to keep the faulter blocked.

Ok, changed the commit that way that such faults don't get answered/replied to anymore and instead invoke the signal handler of the paging object.

skalk, 2018-09-21:
Ok, sel4 seems to _not_ set the strict alignment setting, but clears the corresponding bit in the SCTLR. When I artificially produce a "soft" alignment fault by dereferencing an unaligned pointer, it does not fault on sel4. Unfortunately, if I look at the instruction pointer of the faulting usb_drv, it only does 32-bit word load, which should not fault with the A bit not set...
I have no idea at this point.

skalk, 2018-09-24:
Ok, sel4 seems to _not_ set the strict alignment setting, but clears the corresponding bit in the SCTLR. When I artificially produce a "soft" alignment fault by dereferencing an unaligned pointer, it does not fault on sel4. Unfortunately, if I look at the instruction pointer of the faulting usb_drv, it only does 32-bit word load, which should not fault with the A bit not set...

The difference in between base-hw and sel4 regarding the alignment faults seems to be the paging attributes. I assume that sel4 mark our DMA pages in the usb driver as device memory, whereby base-hw marks it as uncached memory. Refering to device memory the ARM TRM states: "Unaligned accesses are not permitted to these regions even when unaligned access support is enabled."

alex-ab, 2018-09-26:
I mark this issue as fixed, as the alignment fault can be now detected and the mapping loop can be avoided. The actual reason, why the alignment fault happens is another issue if someone wants investigate.

commit 4fe21621 (2018-09-26): sel4/arm: make alignment faults visible

Fixes #2993
commit a8ed11e7 (2018-10-01): sel4/arm: make alignment faults visible

Fixes #2993
closed 2018-10-01 by chelmuth