genodelabs/genode #2994: Cached_fs_rom: aggressive caching hides updates

ehmry, 2018-09-23:
I made the chached_fs_rom very aggressive about caching, ROMs are seldom removed from the cache, but the side-effect is that stopping a subsystem and replacing the files it uses for ROM will not change the ROM content on next load.

nfeske, 2018-09-23:
Wasn't this the reason to introduce the cached_fs_rom in the first place? If one wants to dynamically change the ROM content, the regular fs_rom should be used. In Sculpt, both variants are available. When the dynamic tweaking of depot content is needed, one can host the corresponding archives at `/depot/local/`. All ROMs obtained from this location are served by the dynamic fs_rom.

I wonder, what do you see lacking? Intuitively, changing the cached_fs_rom to become a little more dynamic looks like the wrong direction because it ultimately will not be dynamic enough and thereby behave unpredictable. Let us better draw a clear line between both (dynamic, cached) use cases.

ehmry, 2018-09-24:
I would not want the cached_fs_rom to track changes to files, just that when a ROM is closed by all clients, the ROM is freed and the file is read anew when requested again. I had a problem trying to replace binaries in place in my depot, and it took me serveral tries to realize that the ROM stayed cached.

nfeske, 2018-09-24:
That's actually what I meant with "unpredictable". Whenever a ROM is not used by anyone else, the update would work. But if there is another user, it won't. I would prefer that it never works because the approach to modify the (immutable) depot is wrong in the first place. One should better use the mutual part of the depot (/depot/local) in those cases.

I agree that it might be useful to evict ROMs at some point, in particular to avoid the boundless inflation of RAM usage of the depot_rom server. But the eviction policy is not straight forward. E.g., when restarting the tc-browser, I would like to keep the disk image in RAM. With your policy, the depot would fetch it from disk unless a second tc-browser is running.

ehmry, 2018-09-24:
I will close the issue then, but it would be nice to apply restrictions to the depot from the perspective of `system-noux` by default, so that the uninformed user would not stumble into this loop of trying to update depot files unsuccessfully.

closed 2018-09-24 by ehmry