genodelabs/genode #2997: qt5: cleanly handle case where close event is ignored

NobodyIII, 2018-09-26:
When the close button is clicked on a Qt5 window, its application may choose to ignore the close event, e.g. when a file has been modified but not saved. Currently, when this happens on Genode, the window will remain open, but will become unresponsive, because Qt attempts to resize the window to 0x0.

I have fixed this with commit b74ebdc2b9541ec1a2238ff5767d7ee00d2f44c1.

NobodyIII, 2018-11-16:
Is there anything that needs to be fixed with this commit, or anything I can do to help get this bugfix merged faster?

chelmuth, 2018-11-19:
Until the window seems to be closed forcefully because of the some historical policy. As I'm not fully sure about this policy I'd like to to ask @cproc and @nfeske to shed some light on this.

NobodyIII, 2018-11-20:
> Sorry @NobodyIII <https://github.com/NobodyIII> for the lacking response.
> At first, I was a bit uncertain about the patch so I deferred the review.
> On a closer look, I think the patch is fine. The width/height of 0 is used
> by the wm to signal the closing of a window to its client. The response (or
> lack thereof) is completely up to the client. You are right that there is
> no need for a client (Qt app) to actually apply this size.
>
> I merged your commit to staging. Note that I slightly tweaked your commit
> message to include the issue number and make the first line fit witin 50
> characters (as a matter of good practice). Are you fine with that?

Sounds good. Thanks for reviewing and merging my commit.

commit a387a062 (2018-11-27): qt5: avoid unresponsive window when aborting close

Fixes #2997
commit 1039ef7a (2018-11-27): qt5: avoid unresponsive window when aborting close

Fixes #2997
closed 2018-11-29 by chelmuth