genodelabs/genode #2998: sculpt: acpica launcher fails

chelmuth, 2018-09-26:
Running the acpica launcher on x250 fails with the following log.

```
[runtime -> extract] extracted '/public/chelmuth/bin/x86_64/acpica/2018-09-11.tar.xz' to '/depot/chelmuth/bin/x86_64/acpica'
[runtime -> update -> dynamic] child "extract" exited with exit value 0
[runtime -> update -> manager] installation complete.
[leitzentrale -> manager] chelmuth/pkg/acpica/2018-09-11 incomplete or missing
[runtime] child "acpica"
[runtime]   RAM quota:  4872K
[runtime]   cap quota:  118
[runtime]   ELF binary: acpica
[runtime]   priority:   0
[runtime] Warning: acpica: incomplete environment ROM session (chelmuth/bin/x86_64/acpica/2018-09-11/acpica)
[runtime -> acpica] detected - ec
[runtime -> acpica] Error: AcpiOsReadPciConfiguration unknown device - segment=0 bdf=00:00.0  reg=0xbc width=0x20
[runtime -> acpica] Error: IO_MEM-session creation failed (cap_quota=3, ram_quota=6K, base=0xccd65000, size=0x1000, wc=no)
[runtime -> acpica] Error: Uncaught exception of type 'Genode::Service_denied'
[runtime -> acpica] Warning: abort called - thread: ep
[runtime] child "acpica" exited with exit value 1
[core] Error: I/O memory [00000000ccd65000,00000000ccd66000) not available
[core] Error: Local MMIO mapping failed!
```

Could not reproduce the good case but IIRC it worked in August.

nfeske, 2018-09-26:
I'm not sure if this is related, but on my machine (x260), acpica starts up and repeatedly prints the message "generate reports because of 20 irqs...", but all reports generated by acpica remain empty. I remember that these reports contained actual information (like battery state) in a previous version.

trimpim, 2018-09-27:
On my T460s i have the same output as @nfeske. 
I did not test since I created the package for acpica a while ago. Back then the reports contained reasonable information.

alex-ab, 2018-10-02:
The commits help to let acpica to survive. Still, it is not working properly. After reverting the commits of #2935 i have a to some degree better working version for my machine.

chelmuth, 2022-11-22:
This issue is outdated.

closed 2022-11-22 by chelmuth