genodelabs/genode #2999: Nic_router: deterministic addresses?

ehmry, 2018-09-26:
This is a feature request for deterministic (or highly probabalistic) MAC and IP addresses, I would like my VM to see the same address on each boot, regardless of boot order. I imagine this could be done by passing  session labels through a 32-bit hash function and use the lower bits for addresses. This would require some collision detection, but I would hope that is not much more complicated than the current address allocator.

m-stein, 2018-09-26:
In case of the IP addresses wouldn't it be a solution to configure your VM stacks statically? You would have to use IPs that are not in the range of the corresponding <dhcp-server> tag at the NIC router.

ehmry, 2018-09-26:
My first thought was to fix MAC addresses, not IPs, because my Ubuntu VM tracks interfaces by MAC.

m-stein, 2018-09-26:
I'm not that happy with the approach you suggested. When having such hidden intelligence inside the router, a user could take the assumption, that his MAC magically always remains the same as granted. But at some point in time, this assumption is violated by the NIC router giving the user a hard time to figure out why his client doesn't work anymore.

How about explicit configuration?
```
<config>
   <downlinks>
      <mac label="vm1 -> " address="02:02:02:02:01" />
      <mac label="vm2 -> " address="02:02:02:02:02" />
      <policy label_prefix="vm" domain="vm_subnet" />
   </downlinks>
</config>
```

chelmuth, 2018-09-27:
I'm not decided yet if I like or oppose the ongoing additions of features to nic_router, but if we're going for *per-label address configuration* I'd like to see fixed IP address configuration too.
```
<ip label="vm2 -> " address="10.10.0.2"/>
```
This is much easier to handle than static IP addresses in VMs.

nfeske, 2018-09-27:
Maybe just a silly idea, but what about using the IP address of a client as the last four bytes of its MAC address and pin the first two bytes to a fixed value (e.g., 00:00)?

m-stein, 2018-09-27:
@nfeske Admittedly, I think this is not very practical.
1) There is no relation between a NIC session and an IP peer. Multiple IPs or none might exist behind one client.
2) The MAC must be determined at session creation, IPs come into play a lot later (after DHCP, or even only through packet parsing).
3) With IPv6 we would get into the trouble of combining the suggested method with the config bits in MAC addresses without creating ambiguous addresses.

m-stein, 2018-09-27:
@chelmuth Sorry, I don't understand clearly what you mean with "... using multiple domain configurations for clients that should be configured this way"

I don't like to enhance the feature set of the router neither but I can see the need for static MACs. I'm still thinking about an elegant solution.

ehmry, 2018-09-28:
I still think we can hash session labels down to 16 or 24 bits and not have collisions in practice (deplicate session labels would be hashed repeatedly until a unique result is produced).

m-stein, 2019-03-19:
I re-opened the issue as I think it's not that unimportant and I'd like to give it another try. Offline discussions last time ran off the track a little. After thinking about the topic anew, I found myself enforced in opting for an explicit configuration approach. 

This is why I'd stick to to this approach
* As mentioned earlier, I'm not enthusiastic about adding an approach that uses some kind of hashing because 
  * it might do what I want most of the time but when and how is not transparent to the user and
  * it would make it hard to manage fixed dynamic IPs in bigger scenarios. Imagine a scenario with multiple layers of NIC mutiplexing. In this scenario, the best option to relate IPs to peers is by their MAC, as the session label can no longer identify a peer to upper layers.
* Enhancing the configuration interface is not so cool because it means extra effort for a user, but
  * a complete transparent solution is not in sight and hiding the complexity by using a solution that compensates for a missing user interface with heuristics and non-reliablity also means extra effort for the user
  * a user needs to know about the new interface only if he requires fixed IPs/MACs
* In case the approach we choose now turns out to not be the final solution after some years, IMO an explicitely configured solution would be easier to replace because
  * systems don't rely on it by accident and
  * a configuration that has become invalid is much easier to spot than a changed paragraph in a README.

However, I'd like to propose an configuration interface that slightly differs from my first suggestion:
```
<config>
   <policy label="vm1" domain="vms" first_mac="02:02:02:02:02:01"/>
   <policy label="vm2" domain="vms" first_mac="02:02:02:02:02:02"/>
   <domain name="vms">
      <dhcp-server first_ip="192.168.0.1" last_ip="192.168.0.16">
         <assign mac="02:02:02:02:02:01" ip="192.168.0.1"/>
         <assign mac="02:02:02:02:02:02" ip="192.168.0.2"/>
      </dhcp_server>
   </domain>
</config>
```
This is why I changed my interface proposal:
* Instead of new nodes ``<downlinks><mac><downlinks/>`` I think it is sufficient to integrate it in the existing policy tag. The ``first_mac`` contains the MAC that is given to the first session matching the policy, so, the user can ensure that only one session will match the policy if reliable fixed MACs are desired. This spares additional complexity and doesn't render existing configurations invalid.
* With fixed MACs, it is possible to relate MACs to IPs in the DHCP server. I think it's good to require all fixed IPs to be in the range ``[first_ip, last_ip]`` of the DHCP server and thereby discard the option of a DHCP server without attributes ``first_ip``, ``last_ip`` but only ``<assign>`` sub-tags. This way, it remains very clear which IP addresses are managed by the DHCP server and which not. This interface addition is also backwards compatible.

m-stein, 2019-03-19:
A working prototype of the ``first_mac`` attribute is implemented and tested via the ping_nic_router test:
71407a5e6b wip "nic_router: first_mac attribute in session policies"

m-stein, 2019-03-20:
A working prototype of the ``assign`` tag is implemented and tested via the ping_nic_router test:
e5826acf6a wip "nic_router: <assign> tag for DHCP servers"

m-stein, 2019-03-21:
I just realized that the topic becomes a bit complicated when considering re-configuration. The main problem is that we cannot guarantee for the 'first_mac' attribute to take effect if it was added with a re-configuration. The configured MAC address might legally be allocated already. To regain determinism, we could only pro-actively close the session that holds the MAC address. This is because the user may rely on the fact that no one else uses the MAC address when it comes to deterministic IP assignments. But I don't think it is a good idea to let the server close a session, especially if the user is not capable of foreseeing whether his new configuration will have this effect as he cannot know which MAC addresses are already taken. This is why I would come up with two consecutive implementation steps:
1) Replace 'first_mac' by 'prefer_mac'  which can be ignored by the router (this would cause a warning) and also reflects this through its name, and
2) if we need to have reliably deterministic MACs in the future, introduce a configuration interface that first tries to reserve certain MACs and report the result of this action, so, the user can then refer to definitely available MACs in a second configuration step.

nfeske, 2019-03-21:
In the face of the many topics that are of current interest, I'm wondering why you are pushing this issue right now. Is it on the critical path for anything important?

m-stein, 2019-03-21:
I announced my interest in this issue to all of us this Monday. I re-opened it because two persons were asking for it in the past. I don't know how critical it is for them. But my above post was only the summary of what I found yesterday. Currently I'm actually at other issues and not planned to continue with this this week.

m-stein, 2019-03-21:
I must admit that I'm sometimes a bit confused with the definition of what is generally important and what not. For instance, issue #3237 was considered very useful in the end although it was also only motivated by the personal request of a colleague.

nfeske, 2019-03-21:
Thank you for clarifying. I just mistook your activity in this issue as a sense of urgency but did not know where it came from.

Regarding priorities, I sense #3238 as a pressing (and also long-standing) issue that bothers some of us (who actively monitor the CI results) every day. In contrast, the new NIC router features fall in hobby territory as long as there is neither demand for them by customers nor by our official road map.

m-stein, 2019-03-26:
I'd like to lay this issue to sleep again, but for everyone who wants to have deterministic MACs/IPs I'll keep these two patches online:
e5826acf6a wip "nic_router: \<assign\> tag for DHCP servers"
71407a5e6b wip "nic_router: first_mac attribute in session policies"

They should work just fine as long as you don't touch ``<assign>`` or ``first_ip`` on re-configuration (don't remove, add, or modify them respectively the tags and attributes they depend on).

closed 2019-03-26 by m-stein