genodelabs/genode #3013: Debug: text console in grub on nova causes boot loop

jklmnn, 2018-10-11:
When I enable the text console in Grub2 with Nova, I get stuck in a boot loop. Bender boots but Nova seems to crash and then reboots. This happens also without bender.

This can be reproduced by applying https://github.com/genodelabs/genode/commit/1ae2e1c7c1ac13b574762da7ce0498d82aff6048 to master (as in #2880) and running e.g. run/log on Nova.

jklmnn, 2018-10-11:
This issue has been introduced with https://github.com/genodelabs/genode/commit/3824ad47566fe4331453585ed3db03fa0706cf9f.

alex-ab, 2018-10-11:
Thanks for reporting. Please try c66408d0d7a5e982e7d5b7e44b76ff1706af1650

jklmnn, 2018-10-11:
That fixes it.

chelmuth, 2018-10-17:
Is c66408d the aspired fix?

alex-ab, 2018-10-17:
No, please use 1c569534776277bb10afea4ef7b0672565b3ba40 which is adjusted to apply for staging - thx.

commit cf426e17 (2018-10-17): nova: exclude framebuffer range from buddy memory

Fixes #3013
commit e6ddffb9 (2018-10-29): nova: exclude framebuffer range from buddy memory

Fixes #3013
closed 2018-10-29 by chelmuth