genodelabs/genode #3016: memory performance test

skalk, 2018-10-15:
Recently, we have detected a memory throughput regression on i.MX6 platforms. During investigation we detected that the L2 cache was not enabled on base-hw for that platform anymore. To faster detect those problems a nightly test would be fine.

chelmuth, 2018-10-18:
The test reveals that memcpy on RAM dataspaces is far to slow on base-linux compared to the pure Linux test program. First looks into mmap attributes did not reveal the cause. @cnuke and me speculate that it may be related to anonymous mmap vs. mmap'd files in /tmp.

closed 2018-10-29 by chelmuth

chelmuth, 2018-11-02:
My last remark about base-linux is wrong as long as the same optimization flags are used for linux and Genode. With O2, I get now
```
Genode:
bytewise:                copied 8 GB in 5849 milliseconds (1400 MiB/sec)
memcpy:                  copied 8 GB in 5412 milliseconds (1513 MiB/sec)
memset:                  copied 8 GB in 5471 milliseconds (1497 MiB/sec)
libc memcpy:             copied 8 GB in 2541 milliseconds (3223 MiB/sec)
libc memset:             copied 8 GB in 1828 milliseconds (4481 MiB/sec)
memcpy (uncached write): copied 8 GB in 5913 milliseconds (1385 MiB/sec)
memcpy (uncached read):  copied 8 GB in 5749 milliseconds (1424 MiB/sec)

Linux:
bytewise memcpy test:
copied 8388608 KiB in 5200978 usecs (1575 MiB/sec)
libc memcpy test:
copied 8388608 KiB in 1783476 usecs (4593 MiB/sec)
libc memset test:
copied 8388608 KiB in 1673585 usecs (4894 MiB/sec)
```

chelmuth, 2018-11-02:
And with O3 it just flies...
```
Genode:
bytewise:                copied 8 GB in 2512 milliseconds (3261 MiB/sec)
memcpy:                  copied 8 GB in 2514 milliseconds (3258 MiB/sec)
memset:                  copied 8 GB in 1693 milliseconds (4838 MiB/sec)
libc memcpy:             copied 8 GB in 2492 milliseconds (3287 MiB/sec)
libc memset:             copied 8 GB in 1658 milliseconds (4940 MiB/sec)
memcpy (uncached write): copied 8 GB in 2523 milliseconds (3246 MiB/sec)
memcpy (uncached read):  copied 8 GB in 2574 milliseconds (3182 MiB/sec)

Linux:
 g++ -I.. -O3 -Wall -Wextra -Weffc++ -std=gnu++11 main.cc -o memcpy
bytewise memcpy test:
copied 8388608 KiB in 1788925 usecs (4579 MiB/sec)
```

commit 789df095 (2018-10-17): test for measuring memcpy throughput (fix #3016)
commit 594f215d (2018-10-17): test for measuring memcpy throughput (fix #3016)
commit f2149327 (2018-10-29): test for measuring memcpy throughput (fix #3016)