genodelabs/genode #3020: Remove VFS Insufficient_buffer exception

ehmry, 2018-10-23:
The `Vfs::File_io_service::Insufficient_buffer` exception is not handled consistently across VFS frontends and thus is a source of potential problems. I suggest that this condition be solved by blocking in the VFS plugins until the expected write behavior of the Libc is better defined.

ehmry, 2018-10-23:
I forgot to mention that this exception is not handled at the vfs import plugin. I would rather handle `WRITE_ERR_WOULD_BLOCK` or no blocking error at all rather than handle both an error code and an exception.

Fixed at 83a96eba658f4b54dbd47ad914b936c39fddbfb8.

chelmuth, 2019-03-14:
Seems related to #2954 but I must admit I'm a bit confused of the web of issues connected to the broader topic. @ehmry could you untangle it?

ehmry, 2019-03-14:
Well these look to be the same, so I will close this leave the earlier issue.

closed 2019-03-14 by ehmry

m-stein, 2019-03-14:
Maybe a mind map would make the ongoing VFS work more transparent and understandable to all of us?

ehmry, 2019-03-14:
@m-stein yes, that would be helpful.

m-stein, 2019-03-14:
Maybe you could give [vym](http://ubuntuguide.net/vym-a-mind-mapping-tool-for-ubuntu-linux) a chance? This would perfectly align with the idea of @cnuke to discuss the VFS/libc work at the next staff meeting and we could also incorporate such a map into our task map.