genodelabs/genode #3022: support for clang/llvm static analyzer

alex-ab, 2018-10-25:
https://clang-analyzer.llvm.org

alex-ab, 2018-10-25:
After installing it with a simple ```apt install clang`` scan-build is available.

The tool tries to guess the used compiler, but fails for Genode. So someone has explicitly to specify it,
e.g.:
```
scan-build --use-c++=/usr/local/genode-gcc/bin/genode-x86-c++ --use-cc=/usr/local/genode-gcc/bin/genode-x86-gcc \
make -C build/x86_64 KERNEL=nova core
```
Does just the documentation about this fact is sufficient or do we want to have a convince option in our build system to do the right thing ?

alex-ab, 2018-10-25:
> Does just the documentation about this fact is sufficient or do we want to have a convince option in our build system to do the right thing ?

ecffc1ce178712cba63cd58968e1209e90f38092 implements such a convenience option:

```STATIC_ANALYZE=1 make -C build/x86_64 KERNEL=hw```

chelmuth, 2018-10-29:
I merged the fixes, please rebase your topic branch.

alex-ab, 2018-11-06:
Our depot tools work not well together with scan-build, neither

```scan-build --use-c++=/usr/local/genode-gcc/bin/genode-x86-c++ --use-cc=/usr/local/genode-gcc/bin/genode-x86-gcc  tool/depot/create ... ```
nor
``` STATIC_ANALYZE=1 tool/depot/create ... ``` (with commit 84f3e36cb76f341b15d784a2dd0c53c0b17afbed)

will cause the tool to actually check the code. The reason is that the [scan-build perl script specifically checks for 'make'](https://github.com/llvm-mirror/clang/blob/master/tools/scan-build/bin/scan-build#L1118) as command. The $(MAKE) variable however used by tool/depot/create and friends will be set to /usr/bin/make, which scan-build ignores.

In order to get scan-build to work with our depot cool, one must not call our depot tools directly, but instead as input to a explicit make call, e.g.:

```STATIC_ANALYZE=1 make -f tool/depot/create ...``` (with commit 84f3e36cb76f341b15d784a2dd0c53c0b17afbed)
or
```scan-build --use-c++=/usr/local/genode-gcc/bin/genode-x86-c++ --use-cc=/usr/local/genode-gcc/bin/genode-x86-gcc  make -f tool/depot/create ...```

chelmuth, 2018-11-23:
@alex-ab may I ask you to rebase your topic branch on current staging 889eb647bf8ee646df26bc2cca29fb604785be03. That would simplify to review which follow-up  commits should still be merged.

alex-ab, 2018-11-23:
please use issue_3022 branch

alex-ab, 2018-12-03:
@chelmuth: please consider for review, you seem to missed it

chelmuth, 2018-12-04:
Merged all commits but the one I commented on.

closed 2018-12-12 by alex-ab