genodelabs/genode #3023: libc: mmap() doesn't read whole file if larger than 63K

NobodyIII, 2018-10-29:
When trying to use dynamically loaded plugins in Qt5, I discovered that part of the plugin file appeared to be zeroed. I traced this back to the libc mmap() implementation. If a file is larger than 63K, only the first 63K is loaded into the mmap() buffer, and the rest is zeroed.

I've fixed this with commit 2d5d9a777614f7b5c4c7371facc17bda5088ed47.

NobodyIII, 2018-11-17:
How is this a feature rather than a bug? This issue is of a very similar nature as #2971, except this one fails silently. The mmap() function is supposed to map all of a file, or else fail cleanly and return a null pointer.

chelmuth, 2018-11-19:
I marked this as a feature as support for file mmap was never designed and implemented completely. You're right, that we could have returned an error early but also this was missed back then. @ehmry could you please review @NobodyIII's implementation?

ehmry, 2018-11-19:
Looks good, I edited the patch to use the `size_t` and `ssize_t` type aliases and added an issue tag at f56e7557f4b00aa4b7eb8e42fb98ab17844437d2, @NobodyIII is this ok with you?

NobodyIII, 2018-11-19:
Looks good. I'm looking forward to having this merged.

nfeske, 2018-11-22:
Merged to staging.

commit 2f09aec3 (2018-11-22): libc: read whole file into buffer

Fix #3023
commit e551bc63 (2018-11-27): libc: read whole file into buffer

Fix #3023
commit cde542c3 (2018-11-27): libc: read whole file into buffer

Fix #3023
closed 2018-11-29 by chelmuth