genodelabs/genode #3024: Split window manager into multiple packages

ehmry, 2018-10-29:
The window manager should be split into a wm, decorator, and layouter package. This would allow the decorator and layouter to be replaced without closing and Nitpicker client sessions.

ehmry, 2018-10-29:
Fixed with ee12e5eb28f9db907cadb23d9fa2c95375e2d80b and acc50b52ad7d48607a130ade9808db7053b0df3a. This complicates the Sculpt launcher because three packages need to be added before windows can be placed and drawn.

nfeske, 2018-11-13:
Thanks @ehmry for the first take. I'd like to take over this issue, considering the following additional thoughts:

* The window-manager pkg in Sculpt should be called "wm" (not "window manager") to keep the component small. I expect that many components will depend on it. So keeping it small helps to keep the diagram tidy and neat.

* Currently, however, there already exists a "wm" pkg that includes the wm, the layouter, and the default decorator (motif style). This pkg is quite convenient, e.g., for the various qt-related run scripts. To resolve the conflict with the new designated "wm" pkg that contains only the wm itself, I'd like to rename the current "wm" pkg to "motif_wm", which makes it intuitively clear that it contains a combination of the wm with the motif-style decorator.

* The window manager must keep the roles of the layouter, the decorator, and regular clients apart. To do that, I'd like to introduce the convention that the respective sessions are labeled accordingly: Any client that implements a decorator should be named with "_decorator" as suffix. Any client that implements a layouter should be named with "_layouter" as suffix.

* The pkgs for the existing two decorators will be called "motif_decorator" and "themed_decorator".

ehmry, 2018-11-13:
Sounds good to me. I tried to configure Report request re-labeling rules in with init's top-level service routing and it seems the label handling isn't there yet.

nfeske, 2018-11-16:
Commit https://github.com/nfeske/genode/commit/c176f1d2e749495dd2c6983c49959c09d5788e0c implements the idea. Only tested via `depot_deploy.run` for now.

commit 576a2177 (2018-11-27): sculpt: split window manager into multiple pkgs

This commit moves the window layouter and window decorator into
dedicated packages that can now be combined with the "wm" server at
runtime and restarted/reconfigured/swapped-out independently.

To use the window manager, one must start the 'wm', 'window_layouter',
and one of the 'motif_decorator' or 'themed_decorator' subsystems.

Fixes #3024
commit c6fd0055 (2018-11-27): sculpt: split window manager into multiple pkgs

This commit moves the window layouter and window decorator into
dedicated packages that can now be combined with the "wm" server at
runtime and restarted/reconfigured/swapped-out independently.

To use the window manager, one must start the 'wm', 'window_layouter',
and one of the 'motif_decorator' or 'themed_decorator' subsystems.

Fixes #3024
closed 2018-11-29 by chelmuth