genodelabs/genode #3027: depot_autopilot: evaluate sets of test packages

m-stein, 2018-10-30:


m-stein, 2018-10-30:
My depot_autopilot branch is rebased to recent staging and contains a (pretty large) commit series. The commits are not tagged with the issue number so far and contain not so much documentation yet but it would be worth start testing the state on all platforms:

15afbf8 remove run scripts + autopilot entries of ported tests
8282cf7 depot_autopilot: evaluate sets of test packages
f304b98 libports/recipes: src/ and pkg/test-ada_secondary_stack
acab6ce libports/recipes: src/ and pkg/test-ada
7c2b047 libports/recipes: src/ and pkg/test-stdcxx
98a5bb0 libports/recipes: src/ and pkg/test-rust
c865388 libports/recipes: src/ and pkg/test-libc_pipe
e1389d0 libports/recipes: src/ and pkg/test-libc_vfs* (FIXME: qemu specific config)
96ab9f7 libports/recipes: src/ and pkg/test-libc_getenv
73ba889 libports/recipes: src/ and api/ada
19e1b27 libports/recipes: api/rust
b79e958 os/recipes: raw/, src/, and pkg/test-init
696b1f9 os/recipes: raw/ and pkg/test-part_blk_mbr
9550602 os/recipes: raw/ and pkg/test-part_blk_gpt
c85c940 os/recipes: raw/ and pkg/test-lx_block
0985424 os/recipes: src/ and pkg/test-xml_node
78bdd58 os/recipes: src/ and pkg/test-xml_generator
2bb084d os/recipes: src/ and pkg/test-weak_ptr
cf2ab18 os/recipes: src/ and pkg/test-vfs_stress*
d3886ce os/recipes: src/ and pkg/test-utf8
9a3ffa2 os/recipes: src/ and pkg/test-trace
4fec68a os/recipes: src/ and pkg/test-trace_logger
b57ffe2 os/recipes: src/ and pkg/test-timer
778ab83 os/recipes: src/ and pkg/test-timed_semaphore
6e9be48 os/recipes: src/ and pkg/test-terminal_crosslink
0e6c17c os/recipes: src/ and pkg/test-rom_blk
de85d28 os/recipes: src/ and pkg/test-resource_yield
9988f7d os/recipes: src/ and pkg/test-resource_request
19a9f64 os/recipes: src/ and pkg/test-ram_fs_chunk
a1fa619 os/recipes: src/ and pkg/test-nic_loopback
56a473a os/recipes: src/ and pkg/test-init_loop
de58be5 os/recipes: src/ and pkg/test-fs_packet
1c5291a os/recipes: src/ and pkg/test-fault_detection
d076b41 os/recipes: src/ and pkg/test-dynamic_config
80fc54a os/recipes: src/ and pkg/test-slab
03f5c7c os/recipes: src/ and pkg/test-report_rom
fcfedc6 os/recipes: src/ and pkg/test-clipboard
c03dd07 os/recipes: src/ and pkg/test-blk
b1f9486 os/recipes: src/ and pkg/test-signal
27b8bf2 os/recipes: src/ and pkg/test-rtc
0c40143 os/recipes: pkg/test-read_only_rom (FIXME: success conditions)
e5f1959 os/recipes: pkg/test-rom_filter
f10c400 os/recipes: pkg/test-fs_rom_update_ram
a8e4082 os/recipes: pkg/test-fs_rom_update_fs
6109f9b os/recipes: pkg/test-dynamic_config_loader
8e6e739 os/recipes: pkg/test-dynamic_config_slave
299b402 os/recipes: pkg/test-blk_cache
082e70e os/recipes: pkg/test-fs_log
bf4509d os/recipes: pkg/test-fs_rom_update
09a5a88 os/recipes: src/trace_logger
63a13a6 os/recipes: src/cpu_burner
4dd5a5a os/recipes: src/terminal_crosslink
8c8c9c9 os/recipes: src/test-immutable_rom
1a5118c os/recipes: src/rom_blk
51e75d7 os/recipes: src/nic_loopback
11657f0 os/recipes: src/dummy
f443593 os/recipes: src/rom_logger
e5e00f1 os/recipes: src/rom_to_file
e71748e os/recipes: src/test-bomb
d2c9828 os/recipes: src/fs_log
d58d6c8 os/recipes: src/sequence
add54f5 os/recipes: src/lx_block
537eb2a os/recipes: src/loader
e1c2027 base/recipes: src/ and pkg/test-rm_fault*
7f83851 base/recipes: src/ and pkg/test-synced_interface
875783b base/recipes: src/ and pkg/test-registry
e899085 base/recipes: src/ and pkg/test-reconstructible
07e0feb base/recipes: src/ and pkg/test-new_delete
f32a4a0 base/recipes: src/ and pkg/test-fpu
1314f8a base/recipes: src/ and pkg/test-ds_ownership
857f086 base/recipes: src/ and pkg/test-rm_nested
7b92558 base/recipes: pkg/test-log
5a42633 base/recipes: pkg/test-mmio
af6688b base/recipes: src/test-segfault
9592129 pkg/test-fs_report: add autopilot events
492d988 autopilot.list: add ada_secondary_stack
80f08db autopilot.list: add lx_block
57407dc autopilot.list: add vfs_stress_*
eac8349 autopilot.list: add terminal_crosslink
24c489c autopilot.list: add rtc
e3e2cf6 autopilot.list: remove redundant rom_blk
3da16c6 autopilot.list: add read_only_rom
eaf09e8 autopilot.list: add blk
19b812e autopilot.list: add fs_rom_update*
199814c autopilot.list: add dynamic_config*
920c19a run/rtc: assert spec x86 and !linux
f29ea7d run/fs_packet: raise test timeout
b916894 run/dynamic_config_slave: fix cap quota
420b688 run/dynamic_config_loader: fix cap quota
3cd44e1 run/dynamic_config: fix caps attribute
d529484 run/trace_logger: remove unnecessary stuff
cc13808 run/terminal_crosslink: fix cap quota
947c882 test/ada_exception: remove as it never worked
21eff99 test/xml_node: move to base
15ba40f test/xml_generator: move to base
4771a22 test/weak_ptr: move to os/ as it uses a timer
f124ccb test/libc_vfs_block: add base lib
6a91343 test/libc_vfs: add base lib
f28927a test/libc_pipe: add base lib
1cf5a7f test/stdcxx: add base lib
663545d test/slab: move to os/ as it uses timer driver
7013063 test/init: remove unused _expected_log_message
ff55da2 test-util_mmio: rename test-mmio
6df124b test/util_mmio: simplify success conditions
8488bba test/rm_nested: destroy all objects before finish
ad98da5 ada lib: prepare for creation of packages
0e41a6a base_types.xsd: allow session labels of length 0
28a0125 tool/run/depot: no duplicates in missing archives

m-stein, 2018-11-06:
Here's another commit series that fixes some problems with the first nightly runs of the depot_autopilot, (adds support for hw+arm, foc+arm, fixes general foc problem) and adds further tests:
314d634 run/python: replace via depot_autopilot
b97ed5b run/tcp_bulk_*: replace via depot_autopilot
823b279 run/sequence: replace via depot_autopilot
7c14940 run/pthread: replace via depot_autopilot
488572f run/ldso: replace via depot_autopilot
93f9ab5 run/libc_vfs_counter: replace via depot_autopilot
6b905e6 run/expat: replace via depot_autopilot
ae6b785 run/libc: replace via depot_autopilot
8989606 run/ada_exception: replace via depot_autopilot
2e72372 liports/recipes: api/ and src/python
02bd568 os/recipes: src/nic_bridge
9cc1d50 init: print label on "no route to service" warning
f5adec7 init: show args on "denied env session" error
6dbf702 timeout_types.xsd: allow 0 seconds
a6a5fa0 libports/src/test: add base lib to tests
4bf8bcc run/depot_autopilot: init, depot_query from depots
cc06463 run/depot_autopilot: explicit session routing
9cdbfef depot_autopilot: README file
0fe81e5 depot_autopilot: config.xsd
49df98f depot_autopilot: sanitize ``&amp;`` in LOG pattern
bd8130e base-foc: recipes for ARMv7a platforms
b5b37e8 base-hw: recipes for ARMv7a platforms
399463c Squash "remove run scripts + autopilot entries of ported tests"
8bd2050 Revert "test/ada_exception: remove as it never worked"

m-stein, 2018-11-06:
I moved the to my merge_to_staging branch, sorry for the inconvenience:
bc5c853 run/python: replace via depot_autopilot
67247bc run/tcp_bulk_*: replace via depot_autopilot
172103f run/sequence: replace via depot_autopilot
a541ec9 run/pthread: replace via depot_autopilot
5194033 run/ldso: replace via depot_autopilot
4ccf166 run/libc_vfs_counter: replace via depot_autopilot
dcf12b0 run/expat: replace via depot_autopilot
8f9faab run/libc: replace via depot_autopilot
599bfc3 run/ada_exception: replace via depot_autopilot
48734d5 liports/recipes: api/ and src/python
a218687 os/recipes: src/nic_bridge
1aa19e0 init: print label on "no route to service" warning
1009ec5 init: show args on "denied env session" error
5e8c891 timeout_types.xsd: allow 0 seconds
cd32ebb libports/src/test: add base lib to tests
804b323 run/depot_autopilot: init, depot_query from depots
ff62192 run/depot_autopilot: explicit session routing
d0f238b depot_autopilot: README file
bef4d30 depot_autopilot: config.xsd
c6ba1cf depot_autopilot: sanitize &amp; in LOG pattern
830789a base-foc: recipes for ARMv7a platforms
a09c770 base-hw: recipes for ARMv7a platforms
1491650 Revert "test/ada_exception: remove as it never worked"

chelmuth, 2018-11-06:
The second series misses 399463c ?! Also I'd much appreciate if you stay with the old way of adding new recipes and replacing tests in autopilot.list as fixups to "remove run scripts + autopilot entries of ported tests" to ease the final squashing.

chelmuth, 2018-11-12:
On sel4 and foc we get
```
[2018-11-12 01:14:25] checking configuration syntax
[2018-11-12 01:14:25]   CHECK init
[2018-11-12 01:14:25]   CHECK init -> depot_autopilot
[2018-11-12 01:14:25] .config_0.xml.tmp:29: element previous-results: Schemas validity error : Element 'previous-results', attribute 'time_sec': '277.94699999999995' is not a valid value of the atomic type 'Seconds'.
[2018-11-12 01:14:25] .config_0.xml.tmp:29: element previous-results: Schemas validity error : Element 'previous-results': Character content other than whitespace is not allowed because the content type is 'element-only'.
[2018-11-12 01:14:25] .config_0.xml.tmp fails to validate
[2018-11-12 01:14:25] Error: Invalid XML syntax
```

m-stein, 2018-11-12:
@chelmuth: I reworked it according to your suggestions and added some fixes:
4c84978 gems/recipes: pkg/test-libc_vfs_audit
c61d370 gems/recipes: src/vfs_audit
fa35478 gems/recipes: src/ and pkg/test-magic_ring_buffer
549aa8c gems/recipes: pkg/test-pipe
1a5e5fb libports/recipes: pkg/test-sequence
ce5b869 libports/recipes: pkg/test-libc_vfs_counter
9e174cb libports/recipes: src/ and pkg/test-python
2c2aab5 libports/recipes: src/ and pkg/test-tcp*
b0c5f31 libports/recipes: src/ and pkg/test-pthread
54ed32e libports/recipes: src/ and pkg/test-ldso
60afc40 libports/recipes: src/ and pkg/test-libc_counter
acf9cd3 libports/recipes: src/ and pkg/test-expat
93dd676 libports/recipes: src/ and pkg/test-libc
c88ecda test/python: add base lib
71ca0a7 test/pthread: add base lib
8951bc9 test/ldso: add base lib
f339d44 Fixup "liports/recipes: api/ and src/python" (less lib symbols)
61fb7bc Squash "remove run scripts + autopilot entries of ported tests"
39a57d6 Fixup "remove run scripts + autopilot entries of ported tests" (test-libc_vfs_audit)
12933d6 Fixup "remove run scripts + autopilot entries of ported tests" (test-magic_ring_buffer)
981bf0b Fixup "remove run scripts + autopilot entries of ported tests" (test-pipe)
186aaaf Fixup "remove run scripts + autopilot entries of ported tests" (test-python)
141ef8d Fixup "remove run scripts + autopilot entries of ported tests" (test-tcp*)
e4566b2 Fixup "remove run scripts + autopilot entries of ported tests" (test-sequence)
e31d0f7 Fixup "remove run scripts + autopilot entries of ported tests" (test-pthread)
00aed1b Fixup "remove run scripts + autopilot entries of ported tests" (test-ldso)
22801c4 Fixup "remove run scripts + autopilot entries of ported tests" (test-libc*_counter)
2c4d338 Fixup "remove run scripts + autopilot entries of ported tests" (test-expat)
ddbb097 Fixup "remove run scripts + autopilot entries of ported tests" (test-libc)
b37ab2d Fixup "depot_autopilot: evaluate sets of test packages" (test-libc_vfs_audit)
999644a Fixup "depot_autopilot: evaluate sets of test packages" (test-magic_ring_buffer)
007a9e5 Fixup "depot_autopilot: evaluate sets of test packages" (test-pipe)
5b71d84 Fixup "depot_autopilot: evaluate sets of test packages" (test-python)
b073e78 Fixup "depot_autopilot: evaluate sets of test packages" (test-tcp*)
7d344ca Fixup "depot_autopilot: evaluate sets of test packages" (test-sequence)
abd79f0 Fixup "depot_autopilot: evaluate sets of test packages" (test-pthread)
6a1b61e Fixup "depot_autopilot: evaluate sets of test packages" (test-ldso)
e665637 Fixup "depot_autopilot: evaluate sets of test packages" (test-libc*_counter)
2f794c4 Fixup "depot_autopilot: evaluate sets of test packages" (test-expat)
5b912d5 Fixup "depot_autopilot: evaluate sets of test packages" (test-libc)
3cec3fc Fixup "depot_autopilot: evaluate sets of test packages" (check platform support)
f17fbec Fixup "depot_autopilot: evaluate sets of test packages" (fix <previous-result> time)
0ce9f06 Fixup "depot_autopilot: evaluate sets of test packages" (init, depot_query from depots)
2b642dc Fixup "depot_autopilot: evaluate sets of test packages" (explicit session routing)
0f1f691 Fixup "depot_autopilot: evaluate sets of test packages" (README file)
867d360 Fixup "depot_autopilot: evaluate sets of test packages" (config.xsd)
491d8ac Fixup "depot_autopilot: evaluate sets of test packages" (sanitize &amp; in LOG pattern)
fc9db96 Fixup "run/libc: replace via depot_autopilot" (revert)
a79103e Fixup "run/expat: replace via depot_autopilot" (revert)
254702e Fixup "run/libc_vfs_counter: replace via depot_autopilot" (revert)
6db4f8f Fixup "run/ldso: replace via depot_autopilot" (revert)
132c598 Fixup "run/pthread: replace via depot_autopilot" (revert)
79d838f Fixup "run/sequence: replace via depot_autopilot" (revert)
0b9d327 Fixup "run/tcp_bulk_*: replace via depot_autopilot" (revert)
f789037 Fixup "run/python: replace via depot_autopilot" (revert)
b45959b Fixup "run/depot_autopilot: init, depot_query from depots" (revert)
ddcea97 Fixup "run/depot_autopilot: explicit session routing" (revert)
ceb78d6 Fixup "depot_autopilot: README file" (revert)
67c41e7 Fixup "depot_autopilot: config.xsd" (revert)
0f92a7d Fixup "depot_autopilot: sanitize &amp; in LOG pattern" (revert)

chelmuth, 2018-11-13:
Thanks, I merged the series from m-stein/merge_to_staging until 4c84978cd6 (resp. 9f477c2f08811f607ff18d71ca5c3c547f9888d0).

I noticed you addressed the config.xsd issue. Did you also address the armv6 issue I asked about?

m-stein, 2018-11-13:
Yes, in 3cec3fcb33700540909ad11c289c673ddb71fc96

m-stein, 2018-11-14:
@chelmuth Seems as I have vergeigt it. Here's is a fix:
17deee4 Fixup "depot_autopilot: evaluate sets of test packages" (remove merge artefact)

m-stein, 2018-11-16:
@chelmuth Two more commits that should give us more nightly results:
6f721bb test/rm_nested: fix EP double dissolve
5368173 run/depot_autopilot: quickfix autopilot+foc+x86

m-stein, 2018-11-20:
@chelmuth You were completely right with the missing power-off. Thank you. Some further commits:
e62ab9d run/depot_autopilot: power off before reboot
40b7350 depot_autopilot: show result statistic
8d84952 pkg/test-rm_fault: event "Error: could modify ROM"

nfeske, 2018-11-26:
@m-stein Do you still have the following points in the back of your mind?

- Moving the rtc test out of depot_autopilot
- Removing the sessions to core's I/O services from depot_autopilot

I'd like to add these changes as fixup to "depot_autopilot: evaluate sets of test packages". Would that be possible?

ehmry, 2018-11-26:
Feature request: *depot_user* override 8ffe9f8e0902b230231ee7184dcd87bbf20c73d3

nfeske, 2018-11-26:
@ehmry Good idea. Merged to staging.

m-stein, 2018-11-26:
@nfeske Sorry for the delayed fix:
9568164 Fixup "depot_autopilot: evaluate sets of test packages" (remove RTC test)

nfeske, 2018-11-26:
Thanks @m-stein. I added your fixup to staging.

chelmuth, 2018-11-27:
Oh sorry, I read your patch the wrong way. In fact you must re-add parent-provides specification for IO_PORT as the timer uses I/O ports to drive the x86 PIT on okl4 and sel4 currently.

chelmuth, 2018-11-27:
The results are promising, thanks! But, there is still some noise (i.e., false positives on unsupported platforms) with hw_muen, hw_zynq_qemu, sel4_wand_quad, and foc_panda we should mute.

skalk, 2018-11-28:
There is a structural problem with test-rm_fault currently. That test is dependent on the fact that ist binary is a ROM dataspace, because it tries to modify its text segment. If you do not provide the test-rm_fault binary as boot module but via some other filesystem, it won't be a ROM dataspace but RAM dataspace nowadays.
Before we do not support this, I vote for taking the rm_fault test away from the depot_autopilot again. Do you agree?

m-stein, 2018-11-28:
Oh, yeah sounds, reasonable... but shouldn't the text segment be (executable|read)-only in any case?

m-stein, 2018-11-28:
Ah, sorry didn't read your message carefully enough. So, I'd vote for taking it out temporarily.

nfeske, 2018-11-28:
That's interesting. So the problem can be solved by replacing the fs_rom in depot_autopilot.run by cached_fs_rom? The latter hands out read-only dataspaces.

skalk, 2018-11-28:
@nfeske: probably you are right. Do you think this is the way to go, or shall we take it out and wait until we can "downgrade" RAM to ROM dataspaces?
I've forgot that there is at least rm_fault_no_nox as another affected test, and maybe (I have to investigate) rm_nested too.

nfeske, 2018-11-28:
@skalk the rw-to-ro downgrade mechanism is already in place. It is used by the cached_fs_rom. So I think this is the way to go.

skalk, 2018-11-28:
@nfeske: ah of course, otherwise there would be no way to provide it.

Anyway, I've tested to use the cached_fs_rom. It works to provide the ROM dataspace, so the client is faulting as expected. However, the fault-handler does not receive the fault signal then. 

For this reason I think it is not good to use this test in depot_autopilot in the first place, but maybe as a supplement. I think we want to test the "pure" core service here first, not the managed dataspace used by cached_fs_rom. Propably because of the managed dataspace used here, the fault handler does not receive the fault-signal when writing to the read-only dataspace, am I right?

nfeske, 2018-11-28:
I agree. The reflection of faults for r/o managed dataspaces hasn't been used so far. So it is plausible that there is something missing.

skalk, 2018-11-28:
According to chelmuth's idea, for the moment I will just replace the test-rm_fault binary used here with ld.lib.so, which still is provided by core as a boot module.

skalk, 2018-11-28:
@chelmuth please consider commit 2a7694f for staging

ehmry, 2018-11-29:
I would like to add Solo5 to autopilot, patch at 270e5636c3f05ec55091a70caedb82a591547d90.

skalk, 2018-11-30:
For some reasons the depot_autopilot does not continue correctly when it got forced to reboot. The phenomena seems to be different depending on the kernel and platform. But it seems to always fail to recover.
@m-stein can you have a loo into this?

m-stein, 2018-12-03:
@chelmuth I also thought about this. But I would like to have a bit more.

I think, the variable names should be shorter, like:
env(DEBUG_PKGS) -> $debug_pkgs
env(DEBUG_BUILDS) -> $debug_builds
env(DEBUG_MODEULES) -> $debug_modules

I often want to debug multiple but few packages, for instance, to see if the system stays alive:
DEBUG_PKGS="test-rm_nested test-log"

I think it is not a bad idea to have all three variables available through the command line.

I'm at creating an alternative patch...

m-stein, 2018-12-03:
A fix for the qemu arguments when re-booting:
d96ad1e run/depot_autopilot: reset qemu args correctly

m-stein, 2018-12-04:
@chelmuth @cnuke: This should help to keep the total result number of the nightly depot_autopilot-graph constant:
06f1189 run/depot_autopilot: print nr of tests to run

skalk, 2018-12-04:
@m-stein the reboot still fails at least on real hardware.
The initial timeout is not enough when running depot_autopilot on Qemu/sel4_x86_32

m-stein, 2018-12-04:
@skalk @chelmuth: I wasn't able to reproduce the sel4 timeout but I hope 40 sec are enough. And we need also a fixup for the qemu_args:
d4d4542 run/depot_autopilot: raise initial timeout to 40s
c2fce14 Fixup "run/depot_autopilot: reset qemu args correctly" (consider qemu_args non-existant)

m-stein, 2018-12-04:
A short overview of the current depot_autopilot fails:
```
x86/x86_32/okl4       During test-vfs_stress "Error: Spawned process died unexpectedly"
x86/x86_32/pistachio  During test-vfs_stress "Unhandled exception 0", then "Error: Spawned process died unexpectedly"
x86/x86_32/sel4       During test-vfs_stress "Error: unexpected exception during fault", then "Error: Spawned process died unexpectedly"
x86/x86_64/sel4       During test-vfs_stress "Error: unexpected exception during fault", then "Error: Spawned process died unexpectedly"
wand_quad/sel4        During test-vfs_stress "Error: Spawned process died unexpectedly"
rpi/hw                During first boot "Image too large: increase CONFIG_SYS_BOOTM_LEN"
panda/hw              During test-sequence reboot, then "kernel initializedError: platform rebooted unexpectedly"
panda/foc             During first build "Test requires base-foc kernel archive"
muen/hw_muen          During first build "Test requires base-hw kernel archive"
```

chelmuth, 2018-12-04:
Or as a task list that can be crossed out (already extended)

- x86/x86_32/okl4
  - [ ] During test-vfs_stress "Error: Spawned process died unexpectedly"
- x86/x86_32/pistachio
  - [ ] During test-vfs_stress "Unhandled exception 0", then "Error: Spawned process died unexpectedly"
- x86/x86_32/sel4
  - [ ] During test-vfs_stress "Error: unexpected exception during fault", then "Error: Spawned process died unexpectedly"
  - [ ] During test-vfs_stress "Error: unexpected exception during fault", then "Error: Spawned process died unexpectedly"
- wand_quad/sel4
  - [ ] During test-vfs_stress "Error: Spawned process died unexpectedly"
- rpi/hw
  - [ ] During first boot "Image too large: increase CONFIG_SYS_BOOTM_LEN"
- panda/hw
  - [ ] During test-sequence reboot, then "kernel initializedError: platform rebooted unexpectedly"
- panda/foc
  - [ ] During first build "Test requires base-foc kernel archive"
- muen/hw_muen
  - [ ] During first build "Test requires base-hw kernel archive"
- linux
  - [ ] test-trace_logger times out
- generic
  - [ ] test-stdcxx sometimes fails (false positive on child "test-stdcxx" exited with exit value 0)
  - [ ] test-magic_ring_buffer sometimes fails (false positive on child "test-stdcxx" exited with exit value 0)
  - [ ] test-tcp_bulk_l*ip sometimes fail (false positive on child "test-stdcxx" exited with exit value 0)

m-stein, 2018-12-04:
I'm missing the :fancy!: emoticon...

m-stein, 2018-12-04:
@chelmuth This should fix [qemu x86_64 foc init_loop] and [linux trace_logger]:
814f07d pkg/test-init_loop: raise timeout to 150s
ad6dc57 base-linux: init trace control

m-stein, 2018-12-04:
@chelmuth Support for muen:
107c20c base-hw/recipes: src/base-hw-muen

m-stein, 2018-12-04:
@chelmuth Fix for FOC+Panda:
273c7fb run/depot_autopilot: do not run on foc+panda

m-stein, 2018-12-04:
So far, I couldn't reproduce the test-vfs_stress_ram problem with timeouts < 600 sec, but without other tests. So, with 600 secs and TEST_PKGS="test-vfs_stress_ram" the "Spawned process died unexpectedly" occurs. In this case, we don't even return from run_genode_until. So the kill command shouldn't be the problem. With smaller timeouts, the reboot works as expected (even with ultra).

skalk, 2018-12-06:
@chelmuth @nfeske would you mind to put commit a4928e4 and 9af34a4 on staging? It solves the reboot problem and reduces the test timeouts.

m-stein, 2018-12-07:
@chelmuth This ensures that the depot_autopilot.run script, when exiting, always prints a result overview of the so far available test results, except the Depot Autopilot component has managed to print the result overview up to this point:
3aaf7a2 run/depot_autopilot: always log available results

skalk, 2018-12-10:
I'm afraid there are still some false-positives left, or at least I interpret the following signs as such:
When running test-rm_nested on Fiasco.OC x86 either 32bit or 64bit, the test si timed out and gets rebooted. But Before the timeout triggers, we see these result:
```
[init -> depot_autopilot] 3.490 [init -> test-rm_nested] region-map state is READ_FAULT, pf_addr=0x0
[init -> depot_autopilot] 3.495 [init -> test-rm_nested] allocate dataspace and attach it to sub region map
[init -> depot_autopilot] 3.502 [init -> test-rm_nested] returning from handle_fault
[
```
Be aware of the last sign in the last line. It seems as if the log gets disturbed somehow. Which indicates a problem with the surronding depot_autopilot run-script.

skalk, 2018-12-10:
Another issue regarding the test results occurred on hw_muen:
```
...
[2018-12-10 06:27:06] [init -> depot_autopilot] 0.092 [init -> test-xml_node] [0m[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot] 0.092 [init -> test-xml_node] -- Test parsing XML with comments --[0m[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot] 0.093 [init -> test-xml_node] XML node: name = "config", number of subnodes = 2[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot] 0.094 [init -> test-xml_node]   XML node: name = "visible-tag", leaf content = ""[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot] 0.095 [init -> test-xml_node]   XML node: name = "visible-tag", leaf content = ""[0m
[2018-12-10 06:27:06]    ok         3.035  log "[init -> loader -> test-dynamic_ ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-dynamic_config_slave       ok         2.504  log "[init -> test-dynamic_config_mas ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-expat                      ok         0.053  log "[init -> test-expat]  start of e ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-fault_detection            ok         0.351  log "--- finished fault_detection tes ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-fs_log                     ok         0.569  log "[0] Done."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-fs_packet                  ok        19.621  log "child "short-test" exited with e ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-fs_report                  ok         1.610  log "child "test-fs_report" exited wi ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-fs_rom_update              ok         6.077  log "&lt;config iteration="4" />"[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-fs_rom_update_fs           ok         6.109  log "&lt;config iteration="4" />"[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-fs_rom_update_ram          ok         6.087  log "&lt;config iteration="4" />"[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-init                       ok        20.981  log "child "test-init" exited with ex ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-init_loop                  ok         5.510  log "child "test-init_loop" exited wi ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-ldso                       ok         0.107  log "[init -> test-ldso] Lib_2_global ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-libc                       ok         3.474  log "child "test-libc" exited with ex ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-libc_counter               ok        11.132  log "child "test-libc_counter-sink" e ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-libc_getenv                ok         0.046  log "[init -> test-libc_getenv] foo=" ..."[0m[0m
[2018-12-10 06:27:06] [init -> depot_autopilot]  test-libc_pipe                  ok         0.071  log "child "test-libc_pipe" exited wi ..."[0m[0m
[2018-12-10 06:27:07] [init -> depot_autopilot]  test-libc_vfs                   ok         0.205  log "child "test-libc_vfs" exited wit ..."[0m[0m
...
```
In the middle of the test-xml_node output the overall test-results are printed. The test-xml_node is then marked as failed, although there seems to be no problem with executing the test on this platform solely.

skalk, 2018-12-11:
Update to the rm_nested test on FIasco.OC/x86. It is a real issue and has nothing to do with the depot_autopilot. Above mentioned problem with hw_muen still remains.

skalk, 2018-12-12:
hw/panda also showed a false-positive regarding this test

skalk, 2018-12-12:
@m-stein: The same applies to test-tcp_bulk_lxip under Fiasco.OC/x86_32 where the success string `child "recv" exited with exit value 0` is printed, but it times out anyway.

skalk, 2018-12-12:
@m-stein and `child "test-fs_report" exited with exit value 0` in test-fs_report under Fiasco.OC/x86_64 (both last ones under Qemu)

skalk, 2018-12-12:
@m-stein and `child "test-rm_fault" exited with exit value 0` in test-rm_fault under Sel4/X86_64 (hardware)

m-stein, 2018-12-19:
While debugging the very sporadic false negatives, I came up with this feature:
a8ef3d5 depot_autopilot: provide repeat mode

nfeske, 2019-01-07:
Solved in master. Subsequent topics should go into separate issues.

closed 2019-01-07 by nfeske