genodelabs/genode #3029: File_system::Status: add modification time, access time, and executable bit

NobodyIII, 2018-11-01:
Currently, the File_system interface provides no way to determine a file's modification time, which is problematic for components that use modification times to determine whether something needs to be updated.

I noticed this with QFileSystemWatcher, which uses its polling backend on Genode. I've ported Qt5CT, a theme changer for Qt, and it only partially works. While I can edit the settings, Qt components won't update to use the latest settings, unless the config file didn't exist beforehand.

This will likely also be a problem for Git (not yet ported), which seems to also use modification date to determine whether a file has been modified. Git also uses file sizes, so it will notice modifications more often than QFileSystemWatcher, but will still occasionally miss changes, particularly in port hash files.

Of course, to fix the problem for the cases I've mentioned, the filesystem drivers and servers will need to support it as well, along with the VFS, its plugins, and its libc plugin. I'll create a separate issue for the VFS.

NobodyIII, 2018-11-01:
My inclination is to use Unix timestamps (long) for the access and modification times. This would minimize the translation needed for reading times, as well as minimizing the size of File_system::Status. It would also simplify adding times, which might be helpful at some point.

ehmry, 2018-11-07:
I am not comfortable with adding unix time to the VFS, it would require adding RTC and time tracking code to plugins and adds more problems than it solves. The only solution to the 2038 problem is not to participate. Also, the executable bit serves to prevent execution of arbitrary code by unprivileged users and is only a hassle on single-user systems.

ehmry, 2018-11-07:
To clarify my position on the execute bit, whenever the exec bit and file-system layout conventions become inconsistent, it will cause frustration. Consider adding a directory to the PATH environment of a shell where the exec bit is missing on files, which might happen for a number of reasons. If maintaining an exec bit is necessary for backup purposes, I think   this is best solved in an archiving layer.

NobodyIII, 2018-11-07:
We could solve the 2038 problem by using long long. As for the RTC and time tracking, it would be optional, and would probably be simple and easy in most cases anyway.

As for the execute bit, I suspect that we'd ignore it for the most part anyway, but it could be useful when interacting with other operating systems, e.g. Linux in VirtualBox via a shared folder. I don't care much about it myself, at least for now, but I thought I'd add it. Then again, other systems have rwx flags for user, group, and others, so I'm not sure what makes the most sense here.

chelmuth, 2019-03-14:
Related to #1784.

cnuke, 2019-03-18:
I will close this issue as it is to some degree a duplicate of #1784. For now there are no plans to either implement timestamps besides the modification time and handle the execution bit.

closed 2019-03-18 by cnuke