genodelabs/genode #3031: Flexible window layouter

nfeske, 2018-11-06:
Genode's window-manager architecture consists of separate components for the GUI multiplexing (nitpicker), window layout (layouter), window decorations (decorator), and the management of the interplay between those parts (wm). Until now, the existing layouter implements a simple floating-window policy only. For daily use, its current feature set is too limited. Hence, the existing floating window layouter should be replaced by a more flexible component with the following features:

* Floating windows (like the existing one),
* Static partitioning of the screen for placing panels, overlays, and similar GUI elements,
* Dynamic reconfiguration of the layout policy, and
* A save-state mechanism that remembers window layouts across reboots.

As the ultimate goal, the new window layouter should support a variety of user-interface paradigms such a virtual consoles, virtual desktops, screen tiling, and tabbed windows. However, those paradigms should not be built-in but rather supplemented by external policy components.

Related to #60 and #3024.

maverick74, 2018-11-15:
@nfeske will this allow things like start-menu-like, taskbar, clock and those "regular" things one expect from a regular OS?

nfeske, 2018-11-15:
@maverick74 yes, the layouter will support that in principle, i.e., by preserving a certain screen area for a panel. A clock or other status displays are straight forward to implement because such widgets do not interact with the window environment. But the integration of a taskbar (for certain window-management actions) or a start menu (for launching new subsystems) are not addressed by the layouter directly. There are still some gaps to fill.

maverick74, 2018-11-15:
@nfeske thanks for clarifying. :)

Menu and Taskbar, i believe, are extremely important so that the *community* might recognize the project as a viable alternative to other OS's ( even if this has to be taken with a grain of salt due to the limitations that still exist. But still a good alternative to things like e.g.redoxos ).

That will - possibly - bring in the attention of the broader community and hopefully more developers/contributors.

Would be good to have those basics not to far in time... :)

commit d5dd93a8 (2018-11-08): gems: flexible window layouter

This commit replaces the former floating_window_layouter with a new
window_layouter component that supports the subdivision of screen space
into columns and rows, the concept of layers, and the principle ability
to store window layout information across reboots. The latter is
accomplished by reflecting the component's internal state as a 'rules'
report to the outside.

Fixes #3031
closed 2018-11-16 by chelmuth