genodelabs/genode #3032: RNG service

ehmry, 2018-11-06:
This is not a new idea, but I propose an RNG service similar to the RTC service, a simple session with a single method that returns a small buffer, maybe just a 64-bit integer. Applications that use this service must repeatedly call this method to gather more entropy or use an amplifier and maintain an entropy pool. An implementation of this service could build an entropy pool by hashing system reports and periodically stir in noise from jitterentropy.

I would argue that this is safer than using `/dev/random` because it reduces runtime configuration problems. An application that reads the wrong file can be a fatal mistake, whereas little can go wrong with a library that makes synchronous calls to an RPC service. Support for [arc4random](https://www.freebsd.org/cgi/man.cgi?query=arc4random) could be added to the libc and some crypto libraries could be amended with a `Genode::Env` initializer and use the service directly.

Of course it will still be possible to add VFS plugins that have their own entropy backend or use an RNG service with an amplifier.

nfeske, 2018-11-07:
As discussed offline, I'd like to encourage you to reconsider the creation of a new session interface but use the existing terminal session instead. Please refer to the past discussion on the mailing list (https://lists.genode.org/pipermail/users/2016-November/004845.html) for the rationale behind my opinion.

ehmry, 2018-11-07:
Yes, the Terminal session would work just as well, and would allow the server to return only as much entropy as is available.

closed 2018-11-10 by ehmry