genodelabs/genode #3033: libc/vfs: implement ioctl through a file system

cnuke, 2018-11-07:


cnuke, 2018-11-07:
This commit series was triggered by using a `Terminal` session directly in an libc component as *stdin* and *stdout* (e.g., using irssi via ssh) and touches handling signales in the libc as well as using mutliple files to implement `ioctl()` functionality in the VFS. The last part is realized by introducing a `Device_file_system` specifically for providing a directory that contains all control files along side the device file. The directory is by definition always named after device file, i.e., it is prefixed with a `.`.

NobodyIII, 2018-11-07:
Conceptually, I like this a lot. In particular, this allows the File_system interface to more fully support libc and VFS functionality without increasing its complexity, and allows for easier device/terminal emulation.

As for the implementation, I like it, except it seems that ioctl() would fail if the device or terminal were on a separate VFS server. In os/src/lib/vfs/fs_file_system.h, complete_read() appears to return READ_QUEUED unless queue_read() has been called first and the packets have been received, which would result in ioctl() requests failing. Using read() should usually work, but even read() doesn't guarantee that the whole file will be read into the buffer.

cnuke, 2018-11-08:
Yes, that is one of the reasons why this issue still lacks the fixed label.

cnuke, 2019-10-10:
Superseded by #3519.

closed 2019-10-10 by cnuke