genodelabs/genode #3036: Simplify VFS handle context API

ehmry, 2018-11-12:
The `VFS::Vfs_handle` class contains a definition for a `Context` class, but how this class is used is not documented. Internally this object is attached to I/O response handling lists, but this object can be added and removed from handles arbitrarily by the application. Instead, VFS plugins should only manage lists of handles and handle application contexts set on handles as completely opaque.

A safe, type-checked mechanism for attaching application context to handles should be possible, but I haven't yet found one.

ehmry, 2018-11-12:
VFS API clean-up at bca99d324845cfd77c8f165987d84ce54bec4f1f.

I bring this issue up because I am working with setting Nim application contexts on VFS handles for async I/O.

ehmry, 2018-11-12:
Naming cleanup in FS client plugin at 5af607f92f0d815bf13ac36fb20fa0eb68725abd.

ehmry, 2018-11-13:
I have a patch to merge `Vfs::Io_response_handler`, `Vfs::Watch_response_handler`, and `Vfs::Vfs_handle::Context` into a single per-handle handler object at 1d8e1956e856cd2de9c9a0cbdcf139210c7bc405. This actually simplifies the VFS quite a bit and perhaps the *read_ready* notifications could be merged into this mechanism as well.

ehmry, 2018-11-13:
I say postpone this until after the 18.11 release.

ehmry, 2019-03-11:
I have the next patch for my VFS series at 12adcc9cdb7f184b2a31ce53724827fe8fb6be06. This makes changes to the VFS handles and contexts that is an intermediate step to completely remove the `post_signal_hook` (#3132).

chelmuth, 2019-03-14:
@ehmry as this WiP issue misses the [fixed] tag, please remind me if I miss to review/merge this work after Sculpt CE.

ehmry, 2019-03-19:
12adcc9cdb7f184b2a31ce53724827fe8fb6be06 is ready for staging.

chelmuth, 2019-03-20:
@ehmry 06dc3934f3cb2d1eec87dac183b6f3eeb69803b3 fixes last nights build error. Please check it if this mechanic adaption is semantically correct.

ehmry, 2019-03-20:
@chelmuth, yes, that is correct.

chelmuth, 2019-04-09:
The one (and only) commit referencing this issue entered master. Is there more work pending regarding this issue? In other words: Let's remove the [fixed] tag or close it.

ehmry, 2019-04-10:
Yes, this is finished.

closed 2019-04-10 by ehmry