genodelabs/genode #3040: nic_bridge: fixed MAC addresses

m-stein, 2018-11-14:
Enable configuration of a fixed MAC address for each client.

m-stein, 2018-11-14:
Fixed:
48b61f9 nic_bridge: fixed MAC addresses

@cnuke I'm curious whether your setup now works.

cnuke, 2018-11-14:
@m-stein thanks for the commit, I'll keep you posted.

chelmuth, 2018-11-28:
The commit above breaks the `netperf_l*ip_bridge` tests. @m-stein please investigate.

chelmuth, 2018-11-29:
Reopen because of 

> The commit above breaks the `netperf_l*ip_bridge` tests. @m-stein please investigate.

m-stein, 2018-11-29:
I'm at it.

m-stein, 2018-11-29:
As fixup or selfstanding commit (see commit message):
9c5a967 Fixup "nic_bridge: fixed MAC addresses" (bug when policy missing)

nfeske, 2019-01-31:
I just stumbled upon this issue and see that the patch hasn't been merged yet. I'm wondering, is it related to the noise created by the nic_bridge in our nightly tests?

Now looking at the patch with the background of the API renovation, I propose to change it in the following ways:

* We should demand the definition of a policy. If the user explicitly does not want to express a policy, an empty `<default-policy/>` should be provided. With this in place, the `Session_policy` constructor is never supposed to throw in the good case (when the session becomes established). The exception of the said constructor should not be handled inside the `_create_session` method. It is handled by the root component (`Service_denied`) already. This simplifies the code of the `_create_session` method.
* The `mac` should be obtained from the policy by using `Xml_node::attribute_value`, which alleviates the need to catch any `Nonexistent_attribute` exception.

chelmuth, 2019-01-31:
Isn't the patch merged as 82ded858aae8ca2e13354baa774b9617feccbec3 ?

nfeske, 2019-01-31:
Ok, than I have completely missed the picture. Sorry.

m-stein, 2019-02-06:
@nfeske Your suggestions are something that was already on my todo list. I'll apply them now.

m-stein, 2019-02-06:
How about this:
4ef7caf511 nic_bridge: simplified session request handling
?

nfeske, 2019-02-06:
@m-stein I like it! It looks much more logical now. Thanks for the overhaul.

nfeske, 2019-02-13:
@m-stein I merged your commit to staging.

nfeske, 2019-02-19:
The patch went to master. Unfortunately, there are still several run-script adjustments outstanding. @chelmuth is investigating...

commit b9553c11 (2018-11-16): nic_bridge: fixed MAC addresses

Enable configuration of a fixed MAC address for each client.

Fixes #3040
commit 5c303410 (2018-11-27): nic_bridge: fixed MAC addresses

Enable configuration of a fixed MAC address for each client.

Fixes #3040
commit e0b7fb19 (2018-11-27): nic_bridge: fixed MAC addresses

Enable configuration of a fixed MAC address for each client.

Fixes #3040
closed 2019-02-19 by nfeske