genodelabs/genode #3045: test/rm_nested: uncaught exception

m-stein, 2018-11-15:
There's and uncaught exception ``Signal_receiver::Context_not_associated`` that is thrown because in ``Component::construct`` the test does something like:

1. create entrypoint ``ep``
2. create signal handler ``sigh``, thereby binding it to ``ep``
3. manually call ``ep.dissolve(sigh);``
4. destruct ``sigh``, the destructor unconditionally calls ``ep.dissolve(sigh);``

The manual call to ``dissolve`` is not expected neither by ``Signal_handler`` nor by ``Entrypoint``. So, we can either remove the manual call to ``dissolve`` (the test works without and I don't know what was the reason for doing it in the first place), introduce a ``catch`` statement in the implementation of``Signal_handler`` or ``Entrypoint``, or replace the ``throw`` statement with a warning/error.

m-stein, 2018-11-15:
Any opinion?

nfeske, 2018-11-15:
The dissolve is a legacy from a time when signal contexts were managed manually. The modern `Signal_handler` alleviates the need for manually dissolving the handler. Hence, I'd consider the manual call of `dissolve` as a bug. Let's just remove the call.

m-stein, 2018-11-15:
Ok, thanks for the explanation :)

m-stein, 2018-11-16:
Fixed:
a295a19 test/rm_nested: fix EP double dissolve

commit 52481c85 (2018-11-19): test/rm_nested: fix EP double dissolve

Fixes #3045
commit bbc3dd8e (2018-11-27): test/rm_nested: fix EP double dissolve

Fixes #3045
commit abd38551 (2018-11-27): test/rm_nested: fix EP double dissolve

Fixes #3045
closed 2018-11-29 by chelmuth