genodelabs/genode #3046: qt5: use window title as label for nitpicker session

chelmuth, 2018-11-16:
Enable Qt5 applications to configure the nitpicker session label via `setWindowTitle` from within Qt5 applications, and thus, making them identifiable to other Genode components, like a layout manager. 

Initial patch is 9f6418640c6e954dde1149c129c5c112f793f4fa which lacks sanitizing of the label string (length, invalid characters) currently.

ssumpf, 2018-11-16:
@chelmuth: I assumed this is already handled by `QString` and `windowTitle`

> windowTitle : QString
This property holds the window title (caption)
This property only makes sense for top-level widgets, such as windows and dialogs. If no caption has been set, the title is based of the windowFilePath. If neither of these is set, then the title is an empty string.

chelmuth, 2018-11-19:
Seems my note was too vague. What I meant is that we should apply a reasonable limit to the label-string length to prevent extensive labels and also replace invalid (or unwanted) characters, esp. `"` as the label is an XML attribute.

ssumpf, 2018-11-19:
@chelmuth: I have reworked the commit accordingly to our analysis. Please consider c6081a3f700224b7158d8c36e27768c0d085d396 for staging. Again you might have to update the Qt5 port hash.

nfeske, 2018-11-22:
@ssumpf are you sure that returning the `c_str` pointer is safe? I'm afraid that it may be dangling once the scope of the `label` variable is left. It is probably safer to return a `QString`.

ssumpf, 2018-11-23:
@nfeske: Is 3546131 better?

nfeske, 2018-11-23:
Looks good to me.

ssumpf, 2018-11-23:
@nfeske: There seem to be issues with the removal of MAP_SHARED and qt5_core on staging.

nfeske, 2018-11-23:
I have reverted the MAP_SHARED commit in the meanwhile because it also caused trouble for e2fsprogs.

ssumpf, 2018-11-23:
@nfeske: https://github.com/ssumpf/genode/commit/d0fc62198a37356a99dfee02a1811e3693eb8372 should do the trick.

nfeske, 2018-11-23:
Thanks a lot! Merged to staging.

cproc, 2018-11-27:
Commit 1eee44d creates a numbered dummy label if the window did not have a title.

cproc, 2018-11-27:
Commit 27f8da0 adds a version number to the session label if a window with that label already exists.

commit 6a88f1e7 (2018-11-23): qt5: forward window title to nitpicker

This enables Qt5 applications to set a Genode label via 'setWindowTitle'
from within Qt5 applications, and thus, making them identifiable to
other Genode components, like a layout manager.

fixes #3046
commit 91225fbc (2018-11-29): qt5: forward window title to nitpicker

This enables Qt5 applications to set a Genode label via 'setWindowTitle'
from within Qt5 applications, and thus, making them identifiable to
other Genode components, like a layout manager.

fixes #3046
closed 2018-11-29 by chelmuth