genodelabs/genode #3048: Test code coverage analysis with gcov

cproc, 2018-11-16:
The gcov tool [1] can analyze how often each source code line of a test program got executed.

The usual steps (on POSIX-like systems) to use this tool are:

- When building a test program, special compiler flags add instrumentation code to the program and generate a gcov "note" file with each object file. These note files are needed later on for the final analysis. The program is also linked with the gcov library.

- When the test program is running, it collects coverage data and writes it into a gcov "data" file for each object file when the program exits.

- The 'gcov' program uses the source, note and data files to generate annotated source code files with the number of times each line was executed.

The current implementation for Genode is intended to be used with native (non-libc) Genode tests and the 'depot_autopilot' run script and works like this:

- 'COVERAGE = yes' in the 'target.mk' file of the test program adds the gcov-specific compiler flags and links the program with the gcov library.

- The generated gcov note files get a symlink in the 'build/.../bin/gcov_data' directory, which is added to the binary depot package of the test program, so the note files are available in the depot in the 'depot_autopilot' file system.

- When the test program starts, it calls ```env.exec_static_constructors()``` and each instrumented object file registers itself at the gcov library from a static constructor (auto-generated with the gcov compiler flags).

- The gcov library needs the Genode environment to write the collected data to a file system, so the test program also calls the ```gcov_init()``` function with a reference to the environment.

- When the test program is finished, it calls ```genode_exit()```, which calls an exit handler registered by the gcov library. The gcov library then writes the coverage data files to a RAM file system with a path which matches the path to the note files in the depot in the same file system.

- When all tests are finished, the 'depot_autopilot' runs the 'gcov' program, which scans the file system for the collected gcov data files and writes the final analysis for each file to the log. If the source and api packages are available in the depot, the source code is printed together with the counts and line numbers, else only the counts and line numbers with the text '/\*EOF\*/' on each line are printed.

[1] https://gcc.gnu.org/onlinedocs/gcc/Gcov.html

nfeske, 2018-11-17:
Wow! That's a fantastic line of work and writeup. Thank you @cproc!

Now I suddenly see the the connection with your engagement in #3028. It is so cool to see how the gcov tool fits with @m-stein's depot_autopilot tool just like a glove. :-)

m-stein, 2018-11-17:
@cproc It's cool to see how easy it fits into the depot autopilot runscript. Do you know if there is an easy way to also add the source code of a given package to the depot.tar archive?

cproc, 2018-11-19:
@m-stein: not yet. I'll try to figure something out.

cproc, 2018-11-20:
Commit 6282fcd adds a function to depot.inc which adds the src and api packages of the given pkg or src package paths to the given tar archive.

I adapted the other commits to use this function to add the src and api packages of selected tests to the depot tar.

nfeske, 2018-11-22:
Merged to staging, except for the test-log example. Let us better enable the coverage option of the xml_generator test as an illustration.

nfeske, 2019-01-07:
Fixed in master.

closed 2019-01-07 by nfeske