genodelabs/genode #3049: qt5_component: add libc/posix command line arguments and environment variables

NobodyIII, 2018-11-17:
Currently, unlike posix, qt5_component doesn't read environment variables or command line arguments, which limits both debugging (e.g. QT_DEBUG_PLUGINS can't be set via XML) and usability (e.g. PATH, HOME, XDG_* also can't be set by XML). Command line arguments are used by many different programs that open files, and should also be included.

I have solved this issue by copying code from the posix library into qt5_component in b7546296fdd0d49b0790f74999d1064ed14b62ad. If you want this put in a separate library instead, e.g. qt5_posix, I can make a commit that does that instead.

chelmuth, 2018-11-19:
I marked this as fixed but would prefer to move the utility function that can be used bei qt5, libc, and other Libc components.

cproc, 2020-08-17:
Commit c5806b2 moves the utility function to a header.

nfeske, 2020-08-17:
Very good! Merged to staging.

commit ca5a745d (2020-08-17): qt5_component: support arguments and environment from config

Fixes #3049
closed 2020-08-25 by nfeske