genodelabs/genode #3051: Normalize header directories for contrib code

ehmry, 2018-11-19:
I propose to require that contrib ports put their headers in a top-level include directory named after the port, `include/$PORT`, and all spec headers must be in a top-level spec hierarchy followed by the name of the port such as `include/spec/$SPEC/$PORT` (not `include/$PORT/spec/$SPEC`). This is to make collecting headers for API and source packages easier, and applies to generating an SDK as well.

nfeske, 2018-11-22:
That makes a lot of sense. The original structure was just miming the naming in FreeBSD. Thanks for streamlining it to Genode's conventions. Merged to staging.

cproc, 2018-11-22:
When building the binary packages for 'vfs_jitterentropy' and 'test-ldso', the 'sys/cdefs.h' header is not found anymore.

nfeske, 2018-11-22:
I reverted the commit from staging because I expect build failures at other places, too.

m-stein, 2018-11-22:
@nfeske @cproc Here's a fixup that solves the missing headers:
52bea52 Fixup "Move libc headers to conform to x86_32, x86_64, and arm SPECS"
9e51092 Move libc headers to conform to x86_32, x86_64, and arm SPECS

Furthermore it makes the depot_autopilot happy again :)

nfeske, 2018-11-23:
Thanks @m-stein. I just re-added the commit with your fixup to staging.

chelmuth, 2019-03-14:
The issue is marked as [fixed] and the commit entered master. Can we close it?

ehmry, 2019-03-14:
I think so.

closed 2019-03-14 by ehmry