genodelabs/genode #3052: Remove MAP_SHARED define from libc

ehmry, 2018-11-20:
Using mmap will never work with MAP_SHARED during runtime, so remove its definition to trigger compile-time failures.

ehmry, 2018-11-20:
Patch at f315109526b71f08d07b90cbd40cbffdec8c88d0.

ehmry, 2018-11-20:
Patch with clean diff at b3ec71ced268860bb21478d78715333b2731ba63.

NobodyIII, 2018-11-21:
It looks like MAP_SHARED could be implemented. All it would take is writing the mapped contents back to their original file when msync() is called.

Is there a way to determine which parts (pages, blocks, or something like that) of a dataspace have been modified so we could write those back to the file without having to rewrite the entire file?

ehmry, 2018-11-21:
The file would be lazily mapped or mapped ready-only, and when a page faults occur we would know where the application thinks it can write to the file. Of course if the application were killed unexpectedly, the changes would be lost. It also wouldn't be network transparent, nor would msync be transactional.

NobodyIII, 2018-11-21:
That sounds like the most effective mechanism. Using mmap() with MAP_SHARED seems like a bad way of doing most things, so maybe you're right that it's a good idea to remove MAP_SHARED until we encounter code that needs it.

nfeske, 2018-11-22:
At first I was a bit skeptical about @ehmry's patch. Removing the definition seemed a bit too harsh because I can see that it is not downright impossible to implement the semantics in our libc (at least I have a faint idea).

However, after thinking a bit more about it, I agree with @ehmry because the current version of our libc simply cannot deal with it. It is certainly better to let the attempt to port MAP_SHARED-based software fail earlier (at the compile stage) than later (at runtime).

Once we implement support for MAP_SHARED in our libc (if ever), we can still re-introduce the definition.

nfeske, 2018-11-22:
Unfortunately, this patch no longer applies to staging (after I merged the libc-header commit). @ehmry could you please rebase it?

ehmry, 2018-11-23:
@nfeske see a8b6b41a6d418e0794ff748756021a9ac263e59e

nfeske, 2018-11-23:
Thank you Emery. Merged to staging.

nfeske, 2018-11-23:
Unfortunately, the patch breaks the e2fsprogs noux pkg. I expect that there may be similar trouble ahead with other contrib code. So let us defer the commit for the time after the release.

commit 2893ed24 (2018-11-23): Remove MAP_SHARED define from libc headers

Using mmap will never work with MAP_SHARED during runtime, so remove its
definition to trigger compile-time failures.

Fix #3052
closed 2018-11-26 by ehmry