genodelabs/genode #3054: Merge pthread library into libc

ehmry, 2018-11-21:
Pthreads can be merged into the libc without affecting the semantics of either.

ehmry, 2018-11-21:
Patch at 1ac972a8888e70f66e8ee79f445810aa8fff3aaf.

NobodyIII, 2018-11-21:
My understanding is that pthread is a separate library because Noux doesn't support multithreading within a single process.

ehmry, 2018-11-22:
Noux is working ok, Vbox needs a bit of massaging.

ehmry, 2018-11-22:
I have a fixup for Virtualbox at 465d024fa231f2d605ee6e32e0fcb7444a795c23 on my `issue3054_pthread` branch, the package builds but I do not have time to test it at runtime today.

ehmry, 2018-11-22:
Commit 4faaef2bea93ed5ecf97718fd135e95c6a041ccf restores patches to libc, which are still necessary.

nfeske, 2018-11-22:
Merged to staging.

ehmry, 2018-11-23:
@nfeske fixup for libports at 5544bea1166848f847de04e7c410b2406a457c56

commit 415511d0 (2018-11-22): Merge pthread into libc library

The pthread API is considered a standard feature of libc so better to
simply merge it with the libc. Pthreads are in fact already a part of
the libc in the form of weak symbols. This merger is also a prerequisite
for better integrating pthreads with the libc I/O task.

Fix #3054
commit 7a113841 (2018-11-29): Merge pthread into libc library

The pthread API is considered a standard feature of libc so better to
simply merge it with the libc. Pthreads are in fact already a part of
the libc in the form of weak symbols. This merger is also a prerequisite
for better integrating pthreads with the libc I/O task.

Fix #3054
closed 2018-11-29 by chelmuth