genodelabs/genode #3057: Libc: remove 'clock()'

ehmry, 2018-11-24:
> The clock() function determines the amount	of processor time used since
> the invocation of the calling process, measured in	CLOCKS_PER_SECs	of a
> second.

This procedure is just a shallow wrapper over `getrusage` which we cannot reasonably implement. Whats worse, `clock` is so inaccurate on FreeBSD its not even useful.

> ... the Single UNIX	Specification (``SUSv2'') requires
> CLOCKS_PER_SEC to be defined as one million.  FreeBSD does	not conform to
> this requirement; changing	the value would	introduce binary incompatibil-
> ity and one million is still inadequate on	modern processors.

```c
/* Frequency of the clock ticks reported by clock().  */
#define	CLOCKS_PER_SEC	128

```

ehmry, 2018-11-24:
Commits at 50b7b3ec4d1920eb1e3731ccdf8e685022b70d63 and 7c03ed1beceaca3543a414754852666df7a4c78e.

nfeske, 2018-11-24:
Just out of curiosity, can you share the anecdote that triggered you to remove it?

ehmry, 2018-11-24:
Yes, I was measuring times in the 9P server using the Nim `cpuTim()` procedure, but `cpuTime()` was returning an odd number. This was because the FreeBSD implementation was calling the dummy `getrusage` and returning -1, and Nim was converting -1 to a floating point seconds value. I then reimplemented `clock()` to avoid `getrusage`, but then the result seemed uncomfortably imprecise. That was when I discovered that FreeBSD has kept the unit of time returned by `clock()` at precision of the original implementation for binary compatibility.

https://github.com/nim-lang/Nim/pull/9793

nfeske, 2018-11-24:
I suspected an entertaining story. :-) Thanks for sharing it!

ehmry, 2018-11-26:
I should mention that our current `clock` implementation prints an error about `rusage` then returns -1.

commit 0ddaacf9 (2018-11-29): Remove 'clock()' implementation, print warning

FreeBSD implements 'clock' with an accuracy of 128 ticks-per-second for
compatibility reasons, Linux uses 1000000 per-second. Remove 'clock' and
print an error because it is unlikely that this is the resolution
expected by the application.

Fix #3057
commit 5c2599e2 (2019-01-07): Remove 'clock()' implementation, print warning

FreeBSD implements 'clock' with an accuracy of 128 ticks-per-second for
compatibility reasons, Linux uses 1000000 per-second. Remove 'clock' and
print an error because it is unlikely that this is the resolution
expected by the application.

Fix #3057
closed 2019-01-07 by nfeske