genodelabs/genode #3058: enabling AHCI on AMD Ryzen CPU freezes system on NOVA

NobodyIII, 2018-11-26:
I recently built a new computer with an AMD Ryzen 7 2700X CPU, and I successfully ran Sculpt on it. However, the AHCI driver is disabled except on Intel CPUs, so I saw no SATA block devices. When I ran a modified version of Sculpt that enabled the AHCI driver, the system froze shortly after displaying the menus (including SATA devices) and the log on the screen. I can sometimes move the mouse for a second before this happens, which makes it clearer when the system freezes.
This is different from what happened with my computer that had an AMD FX-8320, with an SB950 southbridge. With the SB950, the AHCI driver blocked forever, waiting for an IRQ that never came, but everything else kept working, making it harmless to enable the driver. On the Ryzen CPU, the AHCI driver gets further, and is able to report all the devices, but it somehow freezes the kernel, or possibly just the USB driver and/or the framebuffer driver.
I'll add a comment once I get a good USB-to-serial adapter, which will help me gather more information.

chelmuth, 2018-11-27:
Note, we do not address AMD-based PC system currently (mostly due lack of time and user/developer/customer interest).

alex-ab, 2018-11-28:
Just as hint, please don't start with some complex, like Sculpt - but instead use some simpler ahci test (ahci_bench or some of the filesystem tests) first.

NobodyIII, 2019-05-02:
My mental health has been declining, and I've been busy with other projects, so I decided to spend $30 to get an NVMe SSD. I probably won't put much focus on this issue anytime soon.

cproc, 2019-11-24:
I could reproduce the problem of 'ahci_block.run' not finishing on an AMD Ryzen system and found that if the 'Cmd::St' register was not set to 1 before writing the 'Sact' register (it should be set to 1 according to the specification), the 'Sact' register was not actually written (this effect did not occur on an Intel system). With commit 80804eb the test succeeded for me, but there might be a better place for calling the 'start()' function.

chelmuth, 2019-11-25:
@ssumpf @cnuke What's your opinion about this patch (esp. regarding the remark about `start()`)?

ssumpf, 2019-11-25:
Since 'Sact' is ncq specific, I think the `start()` is okay there.

cnuke, 2019-11-25:
TBH, I cannot assess the fix but I am eager to try it on my AMD machine :+1:

NobodyIII, 2019-11-26:
It works for me on my Ryzen 7 2700X + AMD B450 machine! Thanks for the fix!
I would love to see this included in the 19.11 release, if possible.

NobodyIII, 2019-11-26:
BTW don't forget to enable AMD AHCI in driver_manager.

cnuke, 2019-11-29:
FWIW, even with this commit AMD's AHCI controller does not work on my x3216 APU (it still does not get the interrupt after executing `identify_device`). But that's okay - the Marvell AHCI controller works.

NobodyIII, 2020-02-28:
non-Intel AHCI is still disabled in driver_manager. This commit fixes that: 1c283ae24f79a4c435979dd536023bbc743fea6e

alex-ab, 2020-03-05:
I tested with commit 1c283ae24f79a4c435979dd536023bbc743fea6e on a machine with the same ahci controller as discussed in #1547. Unfortunately, the Sculpt OS does not boot anymore. Without the commit, Sculpt comes up and is usable at least (usb/ram_fs). Instead of whitelisting everything as done by 1c283ae24f79a4c435979dd536023bbc743fea6e I would prefer to solely to that for controller we are sure about.

NobodyIII, 2020-03-05:
@alex-ab It seems to me like a better approach would be to make Sculpt and
other systems using driver_manager able to tolerate a fault in the AHCI
driver. @nfeske What are your thoughts on this?

nfeske, 2020-03-06:
> What are your thoughts on this?

I share the general sentiment. In practice, however, this becomes very difficult very quickly. Consider the following questions:

- What is a "fault" and how to detect it? Some faults can indeed be detected, e.g., via the heartbeat mechanism. We should try to make use of it. But in the case at hand, there is no page fault but an interrupt storm. That means the AHCI driver lives and responds to I/O. From the heartbeat's perspective it is healthy. This, in turn, raises the question about detecting other means of misbehavior ("too much" CPU load) where it is all too easy to end up with esoteric solutions.

- The AHCI driver seems to successfully report a device and generates a report before entering the bad state. Hence, from the driver manager's point of view, it looks like a device is present. Hence, it considers the device for use. Once selected, there is no sensible policy to deselect it without user intervention because in the normal case, such a policy would not make any sense.

- This brings me to the question about what should happen when a driver misbehaves sometime later. In contrast to a NIC driver, a block driver must never forget any state, neither on disk, nor in the form of DMA buffer content, nor in the payload buffers shared with the client. If the driver dies, there exists no good automatic policy for this case. I presume that the overall goal would be the prevention of data loss (but I may be wrong for certain use cases where liveliness is most important). From this perspective, freezing at that point is actually not too bad. The attempt to restart the driver would probably make things worse because the internal driver state (corresponding loosely the device state) and the content of DMA buffers would get lost, while the client would try to continue working with that mess.

= We would ultimately need to give the user the final word of what to do. This requires user interaction, which moves the problem from the device manager's domain to the sculpt manager's domain.

The bottom line is that I doubt that all this is worth the trouble. The answer should be fixing the driver and/or ensuring that the driver does not touch a device it hasn't been successfully tested with. Given the fact that the driver cannot be expected to work on anything other than Intel, the driver manager's restriction is sensible unless we start taking the responsibility to officially support AMD.

nfeske, 2021-02-26:
There is no work planned to fix this issue.

closed 2021-02-26 by nfeske