genodelabs/genode #3059: Packet stream content accessor is unsafe

ehmry, 2018-11-26:
The `packet_content(Packet_descriptor)` method at packet streams returns nullptr when passed an invalid packet. Unfortunately almost nobody checks this pointer, so I propose to throw an exception in this case and add a safe accessor using a lambda after the next release. This is an issue now because of changes to the packet stream layout, see #3053.

ehmry, 2018-11-26:
Patch 82909ff396dd4cb2d903137107737f46428e993e should be ready for staging. I'm using it now with Sculpt and I'm not having any problems, at least not with the base system servers like nic_router and ahci-fs.

nfeske, 2018-11-26:
The patch looks good, especially since you adapted the affected servers right away. Thank you! I merged it to staging.

ehmry, 2018-11-27:
@chelmuth I have a fixup for USB stuff and Rump_fs at 2a17c27ca389918db532fb81acd5ba86348d7b0d.

ehmry, 2018-11-27:
Its a bit hard to test, because it takes components compiled against different headers to trigger the exception.

chelmuth, 2018-11-27:
Thanks. merged to staging.

ehmry, 2018-11-27:
@chelmuth I'v fixed the issue with run/usb_hid_drv with da2b8663cdf40d2d2734716d55a777a414700768. This avoid exceptions in the case of handling zero-length packets.

ssumpf, 2018-11-27:
@ehmery: The launcher in current Sculpt seems to have problems as well.

ehmry, 2018-11-28:
@ssumpf any good error messages, or just a page fault?

ehmry, 2018-11-28:
@chelmuth fixup for usb_net at 8d7a6b4cef6b8d039cf01f26eadcdd583f307125.

@ssumpf as Joseph pointed out the problem with the launcher is probably a read of a zero length file producing and "invalid" packet, this should be fixed on staging with 61bd23656181a0cf1749c3e202c7c972c51c3e24.

ehmry, 2018-12-02:
I have a commit to add a `packet_content` replacement at 0187b5aaa3332f4aa85ef27b6f82d68fe1228743.

ehmry, 2018-12-14:
I should note that I'm not comfortable replacing everything at once, I think its better to update the *File_system* stuff and try to leave the USB until after we know the *File_system* stuff is working fine.

nfeske, 2020-04-23:
I'm closing this stale issue because there is no plan to pick it up.

closed 2020-04-23 by nfeske