genodelabs/genode #3060: jdk: move to Genode world

ssumpf, 2018-11-26:
* remove JDK from Genode main
* add JDK to Genode world
* enable compiler back end on arm/x86_64 platforms

See also: genodelabs/genode-world#123

nfeske, 2018-11-26:
Merged to staging. Btw, what was your intention behind labeling the issue as "rejected"?

ssumpf, 2018-11-26:
@nfeske: There was simply no right label that said "move to world" or "does not belong here". So I figured to "reject" it here and "feature" it in world. Maybe we can add a label for these cases, which will not be very common of course.

nfeske, 2018-11-26:
Alright. I think the label "cleanup" may be alright in this case. In contrast, the "reject" label is really meant for issues that we won't consider (it refers to the issue itself, not the content of the issue).

commit 04e97dda (2018-11-26): Move OpenJDK to the Genode world repository

This reverts commit 58fb5ed722dc9f1909dc9da35e08ebb788f4ba62.

fixes #3060
closed 2018-11-29 by chelmuth