genodelabs/genode #3061: ada: test for exceptions fails on ARM

skalk, 2018-11-27:
On different ARM platforms and kernels the test `test/ada-exception` fails to catch the exception. It justs aborts. That is why depot_autopilot fails too! I would like to disable the test for ARM until this issue is solved. Any other opinions?

m-stein, 2018-11-27:
I wonder whether ADA exceptions shall be supported on ARM or whether it is only a nice side-effect anyway. Maybe the issue gets out of sight completely once the test is disabled.

@jklmnn What is your opinion?

jklmnn, 2018-11-27:
Well Ada exceptions themselves aren't supported at all. Usually a C++ exception should be thrown (and be catched in your test). I'd like to investigate the issue since Ada exceptions should be catchable at least in C++.

Btw. how do you get log output if you test on a Raspberry Pi (the only ARM device I have ready to use)? Afaik a UART is not available.

jklmnn, 2018-11-27:
If it is time critical for the next release you can also disable the test. But for the long term I'd like to have exception at least some kind of defined behaviour.

chelmuth, 2018-11-27:
What about applying the following diff and running `make -C build/pbxa9 run/depot_autopilot`?

```diff
diff --git a/repos/gems/run/depot_autopilot.run b/repos/gems/run/depot_autopilot.run
index 6ae6610f51..837798c8bf 100644
--- a/repos/gems/run/depot_autopilot.run
+++ b/repos/gems/run/depot_autopilot.run
@@ -21,7 +21,7 @@
 #   set single_test_build { server/ram_fs test/libc_vfs }
 #   set single_test_modules { ram_fs test-libc_vfs vfs.lib.so }
 #
-set single_test_pkg ""
+set single_test_pkg "test-ada_exception"
 set single_test_build { }
 set single_test_modules { }
 
```

jklmnn, 2018-11-27:
Can you please try https://github.com/jklmnn/genode/commit/87925796d9dff096a53745b7d977eda9163b14b8? Exception propagation needs to be enabled for Ada code (and probably should also for C code?). It was just luck that it worked on x86.

chelmuth, 2018-11-28:
This seems to fix the issue, thanks @jklmnn. Regarding your hint of using `-fexceptions` also when compiling C I'm uncertain and have to investigate first.

chelmuth, 2018-11-28:
I checked what assumed up to now and it seems the exception can be thrown successfully through a C function. My test was like follows with `make run/log`.

```diff
diff --git a/repos/base/src/test/log/main.cc b/repos/base/src/test/log/main.cc
index 034f1d09af..056ab6e894 100644
--- a/repos/base/src/test/log/main.cc
+++ b/repos/base/src/test/log/main.cc
@@ -17,6 +17,12 @@
 #include <log_session/connection.h>
 
 
+struct Exc { int a; };
+
+extern "C" void func_throw() { throw Exc{99}; }
+
+extern "C" void func();
+
 void Component::construct(Genode::Env &env)
 {
 	using namespace Genode;
@@ -55,5 +61,9 @@ void Component::construct(Genode::Env &env)
 	buf[2*Log_session::MAX_STRING_LEN - 2] = '\0'; /* end of second line */
 	log(Cstring(buf));
 
+	try { func(); }
+	catch (Exc &e) { log("exc ", e.a); }
+	catch (...) { log("unknown exc"); }
+
 	log("Test done.");
 }
diff --git a/repos/base/src/test/log/module.c b/repos/base/src/test/log/module.c
new file mode 100644
index 0000000000..d97851d2ac
--- /dev/null
+++ b/repos/base/src/test/log/module.c
@@ -0,0 +1,4 @@
+
+void func_throw();
+
+void func() { func_throw(); }
diff --git a/repos/base/src/test/log/target.mk b/repos/base/src/test/log/target.mk
index 9a43e074e7..4e76470066 100644
--- a/repos/base/src/test/log/target.mk
+++ b/repos/base/src/test/log/target.mk
@@ -1,3 +1,4 @@
 TARGET = test-log
 SRC_CC = main.cc
+SRC_C = module.c
 LIBS   = base
```

IMO, this fortunately somewhat contradicts the man page of GCC for `-fexceptions`.

jklmnn, 2018-11-28:
If I understand it correctly disabling `-fexceptions` does not necessarily mean that exceptions are not propagated (e.g. it worked in Ada on x86). And since to my understanding exceptions unwind the stack it depends on how the stack is used in the C/Ada part and how it is structured on the target. So it could work for most cases but fail under certain circumstances.

closed 2018-11-29 by chelmuth