genodelabs/genode #3066: Add stack protector support to base library (but do not enable)

ehmry, 2018-11-30:
I suggest support for GCC's stack protection be added to the base library, this would simply be a location for the stack canary check value `__stack_chk_guard` and an implentation of `__stack_chk_fail` to log an error and prevent the stack from unwinding. This would allow Genode components to be built with stack protection enabled the feature itself would not be enabled at runtime. That is unless the `__stack_chk_guard` were initialized with a nonce.

ehmry, 2018-12-05:
I've rebased and added a test to check that components can use `-fstack-protector`, see 7843d6ddcca8c480bf18a44631118f348d8cf514 and 1f1fcf27c2d0796228b15216cbb408e9f8c19e99.

chelmuth, 2019-03-04:
I rebased your commits incl. a small fixup. Please also have an eye on this issue as I plan to merge it after Sculpt CE.

chelmuth, 2019-03-20:
@ehmry Last nightly shows that stack-protector does not build for x86_32. Please look into this. I already tried to add the symbol to *symbols/ld* but this is insufficient.

```
> make -C build/x86_32 test/stack_smash
checking library dependencies...
  Library ldso-startup
  Library ld
    CONVERT  ld.symbols.s
    ASSEMBLE ld.symbols.o
    MERGE    ld.abi.so
  Library base
  Program test/stack_smash/test-stack_smash
    COMPILE  main.o
    LINK     test-stack_smash
main.o: In function `Component::construct(Genode::Env&)':
.../repos/base/src/test/stack_smash/main.cc:23: undefined reference to `__stack_chk_fail_local'
/usr/local/genode-gcc-17.05/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: test-stack_smash: hidden symbol `__stack_chk_fail_local' isn't defined
/usr/local/genode-gcc-17.05/bin/../lib/gcc/x86_64-pc-elf/6.3.0/../../../../x86_64-pc-elf/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
.../repos/base/mk/prg.mk:212: recipe for target 'test-stack_smash' failed
make[2]: *** [test-stack_smash] Error 1
var/libdeps:79: recipe for target 'test-stack_smash.prg' failed
make[1]: *** [test-stack_smash.prg] Error 2
Makefile:282: recipe for target 'gen_deps_and_build_targets' failed
make: *** [gen_deps_and_build_targets] Error 2
```

chelmuth, 2019-03-20:
@ehmry okay it took me not the whole night to identify a quickfix for our problem which is commit a448848595fac7642724d373873fef7fffc7ae7c and fixes the test in all cases, for example

```
make -C build/pbxa9/ run/test PKG=test-stack_smash KERNEL=hw
make -C build/x86_64 run/test PKG=test-stack_smash KERNEL=linux
make -C build/x86_32 run/test PKG=test-stack_smash KERNEL=linux 
```

ehmry, 2019-03-20:
I found this note in the GCC sources:

```c
/* For 32-bit code we can save PIC register setup by using
   __stack_chk_fail_local hidden function instead of calling
   __stack_chk_fail directly.  64-bit code doesn't need to setup any PIC
   register, so it is better to call __stack_chk_fail directly.  */

static tree ATTRIBUTE_UNUSED
ix86_stack_protect_fail (void)
{
  return TARGET_64BIT
	 ? default_external_stack_protect_fail ()
	 : default_hidden_stack_protect_fail ();
}
```

chelmuth, 2019-03-20:
I found it too and therefore added the reference *https://raw.githubusercontent.com/gcc-mirror/gcc/master/gcc/config/i386/i386.c line 45261* to the `FIXME` comment ;-)

commit 8868839a (2019-03-19): Add stack protector storage to base library

This patch adds the items necessary for building Genode components with
stack protection enabled, but it is not initialized at runtime. They are
provided at the moment as a convenience and do not implement a security
feature.

Fix #3066
commit 6819c43a (2019-04-01): Add stack protector storage to base library

This patch adds the items necessary for building Genode components with
stack protection enabled, but it is not initialized at runtime. They are
provided at the moment as a convenience and do not implement a security
feature.

Fix #3066
closed 2019-04-02 by nfeske