genodelabs/genode #3068: libgmp on ARM fails at runtime

skalk, 2018-12-03:
When using libgmp (used by g++) while running the `noux_tool_chain_auto.run` script on any ARM platform, it fails at runtime due to a missing symbol:
```
[init -> noux -> /libexec/gcc/arm-none-eabi/6.3.0/cc1plus] Error: LD: jump slot relocation failed for symbol: '__gmpn_invert_limb'
[init -> noux -> /libexec/gcc/arm-none-eabi/6.3.0/cc1plus] Error: Uncaught exception of type 'Linker::Not_found'
[init -> noux -> /libexec/gcc/arm-none-eabi/6.3.0/cc1plus] Warning: abort called - thread: ep
[init -> noux] child /libexec/gcc/arm-none-eabi/6.3.0/cc1plus exited with exit value 1
```

cproc, 2018-12-11:
Fixed in 99fe5f3.

nfeske, 2018-12-11:
Thanks. I merged your fix to staging.

commit 44ef8262 (2018-12-11): gmp: search headers in REP_DIR first

Fixes #3068
closed 2019-01-07 by nfeske