genodelabs/genode #3072: Add support for UndefinedBehaviorSanitizer

cproc, 2018-12-06:
The UndefinedBehaviorSanitizer (UBSan) [1][2] can detect undefined behavior while a program is running. It can be used with Genode tests in the 'depot autopilot' scenario in the following way:

- 'SANITIZE_UNDEFINED = yes' in the 'target.mk' file of a test program adds the '-fsanitize=undefined' compiler flag and links the program with the 'libubsan' and 'libsanitizer_common' libraries.
- When the test program starts, it calls ```env.exec_static_constructors()``` and ```sanitizer_init(env)``` to initialize the sanitizer libraries.
- When undefined behavior is detected while the test program is running, a "runtime error: " message including the source code location of the error is printed to the log.

I created a test depot package for the slightly modified 'segfault' test for demonstration. The sanitizer detects and reports the null pointer access at runtime before it happens:

```
[init -> test-segfault] /.../src/test/segfault/main.cc:21:7: runtime error: store to null pointer of type 'int'
```

[1] https://developers.redhat.com/blog/2014/10/16/gcc-undefined-behavior-sanitizer-ubsan/
[2] https://gcc.gnu.org/onlinedocs/gcc/Instrumentation-Options.html

cproc, 2018-12-06:
I created a dedicated sanitizer test now, since the 'segfault' test was already used for the 'rm_detection' test scenario.

nfeske, 2019-01-07:
Merged into master.

closed 2019-01-07 by nfeske