genodelabs/genode #3078: Utility for serving a static ROM session

ehmry, 2018-12-10:
I would like to have a simplier version of the `Dynamic_rom_session` class that takes is its content during construction. Otherwise one has to define a `Content_producer` subclass or use the deprecated `Child_policy_dynamic_rom_file` class instead.

ehmry, 2018-12-10:
Patch at ab9b71480996f1e5764c76a8b88d242c43f62549. I've refactored the *sequence* component to use this so its covered by the autopilot. I also have a Nim library for spawning temporary children that uses this.

nfeske, 2018-12-11:
@ehmry How do you think about replacing the (pointer, size) constructor arguments with an `Xml_node`? It would please my eye much more than the use of pointers and avoids the case where a nullptr is passed to memcpy. The config accessor might then become:

```
return _start_node.has_sub_node("config") ? _start_node.sub_node("config")
                                          : Xml_node("<config/>");
```

ehmry, 2018-12-11:
Sure, I have a fixup at 1d37936474e5f1ad51e027c47498b50a482a6ec9.

nfeske, 2018-12-11:
Great! Do you see a practical use case for the other constructor? I.e., do you need to deliver non-XML content with this mechanism? If there is no immediate use case, I would prefer to drop the (pointer,size) constructor. It can still be added later, should it be really needed.

ehmry, 2018-12-11:
Yes, I am generating XML with Nim and then passing that by pointer, and I want to avoid the intermediate object.

nfeske, 2018-12-11:
Alright. Thank you for the background info. If we keep both constructors, let us implement the Xml_node one as

```
Static_rom_session(Rpc_entrypoint &ep, Pd_session &pd, Region_map &rm,
                   Xml_node const &content)
: Static_rom_session(ep, pd, rm, content.addr(), content.size()) { }
```

...to avoid code duplication and to shave off some lines of code. Please also consider adjusting the comments.

ehmry, 2018-12-11:
I rebased a new commit at 30976d1d6f96ccbccf21863f7b646ac975cf4079 with the recommended changes.

nfeske, 2018-12-12:
Thank you Emery. The `Static_rom_session` looks good now. So the first part of your patch could be merged. Would you consider splitting the commit into two?

The change of the sequence tool introduces a problem though. In the case where no <config> node is present in the <start> node, the original version of the sequence tool forwards the session request to the parent. So it can be redirected to a dedicated ROM. The routing decision is based on the value of `_have_config`. In contrast, the new version unconditionally delivers the static ROM session, which contains the empty <config/> node in this case. The `_have_config` member is still present in the new version but no longer used anywhere.

The patch should either preserve the original behavior or provide an alternative way for supplying non-inline configurations to the children of the sequence tool. E.g., a <start> node could feature an attribute like `config="fetchurl.config"`, which would tell the sequence tool to route the child's config ROM request to the parent with the label rewritten to "fetchurl.config". I would actually prefer this explicit approach to the current (rather opaque) behavior. As of now, your commit would break the genodians.org scenario (https://github.com/genodelabs/genodians.org).

ehmry, 2018-12-14:
Then I would defer this issue until *sequence* can route child configs independently rather than cause a regression.

closed 2018-12-14 by ehmry