genodelabs/genode #3080: trace: test-trace_logger fails on foc_pbxa9

skalk, 2018-12-10:
The condition the test has to achieve is never met on this platform. The "state" of the first thread always stays to be "UNTRACED" and never switches to "TRACED".

skalk, 2018-12-11:
Is this a duplicate of issue #2809? @m-stein what went out of your investigation?

m-stein, 2018-12-17:
This was the outcome, so far:
ad6dc57 base-linux: init trace control

But its not the same with the other platforms. I'm at it.

nfeske, 2018-12-17:
@m-stein can you point me to the branch where ad6dc57179de332a813984d74bd6cb1929e08510 can be cherry-picked from?

m-stein, 2018-12-17:
It's on staging:
af12ba6 base-linux: init trace control

m-stein, 2018-12-17:
The problem apparently is the call of the ASM instruction in Trace::timestamp. The tracing itself works fine if I take out the timestamp in the test client. Otherwise the client gets stuck when calling the instruction without any hint in the log and never reaches the call of Thread::trace. It seems that Trace::timestamp never worked (I checked 18.08/05/02), but then I wonder why the timeout framework works.

m-stein, 2018-12-17:
We can fix this by taking out the timestamp (it is not really needed for the test):
2f1959d test/trace_logger: don't use Trace::timestamp

nfeske, 2018-12-18:
I added https://github.com/genodelabs/genode/commit/a1660268bbef6102ad00bd1da2f5fcdd056818a9 to staging.

nfeske, 2018-12-18:
> I wonder why the timeout framework works.

Likewise.

m-stein, 2018-12-18:
With the patch, test-trace_logger works also for okl4, sel4, fiasco and others that seemed to be broken beforehand on my machine.

The timeout test works for all these platforms but if I add a call to Trace::timestamp, it gets stuck at this call...

chelmuth, 2019-03-04:
The test now succeeds.

closed 2019-03-04 by chelmuth