genodelabs/genode #3081: hw: heavy time-drift on panda and i.MX53 under load

skalk, 2018-12-10:
When running the test-signal on hw_panda or hw_imx53qsb it reboots after a timeout of 70 seconds, although the software stack is still alive and the depot_autopilot component fails to timeout, because its "clock" is heavily slower (measured ~20 seconds when actually 70 seconds passed by, measured on pandaboard).
Interestingly the timer_accuracy test finishs successfully on both platforms. 

Either the very special scheduling scheme of base-hw, the timeout framework here, or the usage of time in the depot_autopilot seems to fail here.

chelmuth, 2018-12-19:
hw_arndale also shows this strange behavior in test-timer.

```
[2018-12-19 02:21:23] [init -> depot_autopilot] --- Run "test-timer" (max 30 sec) ---
[2018-12-19 02:21:23] [init -> depot_autopilot] 
[2018-12-19 02:21:23] [init -> depot_autopilot] 0.077 [init -> client] --- timer test ---
[2018-12-19 02:21:23] [init -> depot_autopilot] 0.095 [init -> client] register two-seconds timeout...
[2018-12-19 02:21:25] [init -> depot_autopilot] 2.097 [init -> client] timeout fired - 0/80/80
[2018-12-19 02:21:28] [init -> depot_autopilot] 3.236 [init -> client] wait 1/10
[2018-12-19 02:21:32] [init -> depot_autopilot] 4.240 [init -> client] wait 2/10
[2018-12-19 02:21:35] [init -> depot_autopilot] 5.244 [init -> client] wait 3/10
[2018-12-19 02:21:39] [init -> depot_autopilot] 6.248 [init -> client] wait 4/10
[2018-12-19 02:21:42] [init -> depot_autopilot] 7.216 [init -> client] wait 5/10
[2018-12-19 02:21:45] [init -> depot_autopilot] 8.286 [init -> client] wait 6/10
[2018-12-19 02:21:49] [init -> depot_autopilot] 9.311 [init -> client] wait 7/10
[2018-12-19 02:21:52] [init -> depot_autopilot] 10.273 [init -> client] wait 8/10
[2018-12-19 02:21:56] [init -> depot_autopilot] 11.303 [init -> client] wait 9/10
[2018-12-19 02:21:59] [init -> depot_autopilot] 12.316 [init -> client] wait 10/10
[2018-12-19 02:22:03] [init ->Error: Test execution timed out
```

ValiValpas, 2019-02-14:
@skalk: If I understand your comments in #3157 correctly, you identified what's causing the time drift already. I assume it is because the timer is automatically reloaded, thus the update_time() method calculates the elapsed time wrongly. Nevertheless, I noticed that there is another (minor) flaw in the time accounting that you might want to consider as well. It results from the way how base-hw uses one-shot timers:

Every time a timeout occurs, the time spent in kernel (until the timer is configured again) remains unaccounted. On my Zynq-7000, this can take up to 12 microseconds. With a fill length of 10ms, you should have a time drift of up to ~0.1% in the long term. Fortunately, I believe, this is acceptable for most applications.

However, this could be elminated by using an incrementing timer and compare registers instead of a decrementing one-shot timer. On modern ARM SoCs, there is either the global timer or the generic timer available. Both support this mode.

skalk, 2019-02-15:
@ValiValpas Thank you for your insights. You are right. I came to the same conclusion when reviewing the kernel timing within the last days. I have already a working solution for Cortex A9 and imx series, but have to change Exynos and x86_64 too. Finally it will hit staging before the February release.
There are even more changes in the timeout framework and kernel timing as outcome, which significantly improves timing on hw.

ValiValpas, 2019-02-15:
@skalk Great. Can't wait to do some experiments with the February release.

skalk, 2019-03-05:
@ValiValpas unfortunately I did not managed to get the timing fixes ready for staging in time to be part of the february release, but if you like you might give commit 8a2b4656ea3b8dad1ee85e3279b15ed12d586d8f and 6c0b37b a try.

skalk, 2019-03-05:
@chelmuth @nfeske: there is still an issue with timing on 32-bit platforms when time passes more than one hour, but this is a limitation of the current timeout framework, I will document extra. Apart from that above commit is ready for staging

chelmuth, 2019-03-05:
Merged to staging.

commit 4f77bcb6 (2019-03-05): hw: increase timing accuracy of kernel (fix #3081)

* Introduce 64-bit tick counter
* Let the timer always count when possible, also if it already fired
* Simplify the kernel syscall API to have one current time call,
  which returns the elapsed microseconds since boot
commit 1c957cc9 (2019-03-07): fixup "hw: increase timing accuracy of kernel (fix #3081)"
commit c9edfee3 (2019-03-07): fixup "hw: increase timing accuracy of kernel (fix #3081)"
commit 4bb64313 (2019-03-07): hw: increase timing accuracy of kernel (fix #3081)

* Introduce 64-bit tick counter
* Let the timer always count when possible, also if it already fired
* Simplify the kernel syscall API to have one current time call,
  which returns the elapsed microseconds since boot
closed 2019-03-19 by chelmuth