genodelabs/genode #3082: test-rm_nested: core hangs on Fiasco.OC/x86*

skalk, 2018-12-11:
When running the test-rm_nested package on real x86 hardware (32/64) with Fiasco.OC, core hangs at some point:
```
...
[init -> depot_autopilot] 3.158 [init -> test-rm_nested] allocate dataspace and attach it to sub region map
[init -> depot_autopilot] 3.171 [init -> test-rm_nested] returning from handle_fault
[init -> depot_autopilot] 3.182 [init -> test-rm_nested] write to 0x2f100
[init -> depot_autopilot] 3.192 [init -> test-rm_nested] write to 0x2f200
[init -> depot_autopilot] 3.201 [init -> test-rm_nested] write to 0x2f300
[init -> depot_autopilot] 3.210 [init -> test-rm_nested] write to 0x2f400
[init -> depot_autopilot] 3.219 [init -> test-rm_nested] write to 0x2f500
[init -> depot_autopilot] 3.229 [init -> test-rm_nested] write to 0x2f600
[init -> depot_autopilot] 3.238 [init -> test-rm_nested] write to 0x2f700
[init -> depot_autopilot] 3.247 [init -> test-rm_nested] write to 0x2f800
[init -> depot_autopilot] 3.256 [init -> test-rm_nested] write to 0x2f900
[init -> depot_autopilot] 3.265 [init -> test-rm_nested] write to 0x2fa00
[init -> depot_autopilot] 3.275 [init -> test-rm_nested] write to 0x2fb00
[init -> depot_autopilot] 3.284 [init -> test-rm_nested] write to 0x2fc00
[init -> depot_autopilot] 3.293 [init -> test-rm_nested] write to 0x2fd00
[init -> depot_autopilot] 3.302 [init -> test-rm_nested] write to 0x2fe00
[init -> depot_autopilot] 3.312 [init -> test-rm_nested] write to 0x2ff00
[init -> depot_autopilot] 3.321 [init -> test-rm_nested] test destruction of region_map
[init -> depot_autopilot] 3.341 [init -> test-rm_nested] test multiple nested regions stacked
[init -> depot_autopilot] 3.355 [init -> test-rm_nested]  region top        [0b000000,0b00a000)
[init -> depot_autopilot] 3.367 [init -> test-rm_nested]  region shim       [0b000000,0b006000) [0b006000,0b00a000)
[init -> depot_autopilot] 3.381 [init -> test-rm_nested]   attached mem                         [0b007000,0b009000)
[init -> depot_autopilot] 3.396 [init -> test-rm_nested]   read     mem                         [0b007000,0b009000) value=0x0
[init -> depot_autopilot] 3.411 [init -> test-rm_nested]    bottom shim (r) [0b000000,0b00a000)
[init -> depot_autopilot] 3.423 [init -> test-rm_nested]    bottom shim (s) [0b000000,0b006000)
[init -> depot_autopilot] 3.436 [init -> test-rm_nested]     attached mem   [0b000000,0b006000)
[init -> depot_autopilot] 3.448 [init -> test-rm_nested]     wrote    mem   [0b000000,0b006000) with value=0xffffffff
[init -> depot_autopilot] 3.462 [init -> test-rm_nested]  try reading mem  0xb007000 - should succeed
[init -> depot_autopilot] 3.475 [init -> test-rm_nested]  try reading mem  0xb008000 - should succeed

KERNEL: Warning: nothing mapped: (Obj_space) from [0xffde5f4c/a]: 00000525 size: 00000001 to [0xffc7d39c/328]

KERNEL: Warning: nothing mapped: (Obj_space) from [0xffde5f4c/a]: 00000525 size: 00000001 to [0xffc7d39c/328]
[init -> depot_autopilot] 3.515 [init -> test-rm_nested] fault handler: waiting for fault signal
[init -> depot_autopilot] 3.528 [init -> test-rm_nested]  try reading mem  0xb006000 - should fail
Warning: page-fault, pager_object: pd='init -> dynamic -> test-rm_nested -> test-rm_nested' thread='ep' ip=0x10054ef pf-addr=0xb006000
[init -> depot_autopilot] 3.529 [init -> test-rm_nested] region-map state is READ_FAULT, pf_addr=0x0
[init -> depot_autopilot] 3.541 [init -> test-rm_nested] allocate dataspace and attach it to sub region map
[init -> depot_autopilot] 3.554 [init -> test-rm_nested] returning from handle_fault
[
```
The kernel is still alive. One can enter the kernel debugger, which shows the following thread list:
```
   34   0     ioport           80      a     -       rcv_wait
   30   0     irq              80      a     -       rcv_wait
   2d   0     pager_ep         80      a     -       rcv_wait
   26   0     ep               80      a    28*      rcv_wait
   20   0     signal_entrypoi  80      a     -       rcv_wait
   11   0     core.pager       80      a     d*      rcv_wait
    e   0     entrypoint       80      a    11       rcv_wait
    b   0     core.main        80      a    26       rcv_wait
    9   0     sigma0           ff      8     -       rcv_wait
    7   1     -----             0      1             ready
    6         -----             0      1             ready
```
As one can see, core hangs in a page-fault.

skalk, 2018-12-11:
Further investigation showed that in generic core code the `unmaped_managed` function calls the `Address_space::flush` Fiasco.OC specific implementation with a core-local address of 0x0 and a size of 6 pages. Therefore, core unmaps all maped items of its first 6 pages. Most of them are not used, but at address 0x5000 is a mapping from the RAM used for core's log buffer that might gets exported as ROM. The unmap of the log buffer leads to the page-fault.

skalk, 2018-12-21:
@chelmuth @nfeske I've reworked the commit 2a12c93 so that it now works including the thread-test. Moreover, I've added a small fix for Fiasco.OC (commit ffd1a91), which warned about a missing capability mapping whenever a thread was started.

chelmuth, 2018-12-22:
I fast-forwarded staging to your merge_to_staging branch. Expect first results beginning of January after the holidays.

commit 1b7fabd4 (2018-12-17): core: support unmap of managed dataspace generally

This commit solves several issues:

* correct calculation of overlap region when detaching regions
  in managed dataspaces
* prevent unmap of Fiasco.OC's core log buffer
* calculate the core-local address of regions in managed dataspaces
  if possible at all and use it to unmap on kernels where this is
  needed

Fix #976
Fix #3082
commit 2a12c93d (2018-12-22): core: support unmap of managed dataspace generally

This commit solves several issues:

* correct calculation of overlap region when detaching regions
  in managed dataspaces
* prevent unmap of Fiasco.OC's core log buffer
* calculate the core-local address of regions in managed dataspaces
  if possible at all and use it to unmap on kernels where this is
  needed

Fix #976
Fix #3082
commit 19fd2b32 (2019-01-04): core: support unmap of managed dataspace generally

This commit solves several issues:

* correct calculation of overlap region when detaching regions
  in managed dataspaces
* prevent unmap of Fiasco.OC's core log buffer
* calculate the core-local address of regions in managed dataspaces
  if possible at all and use it to unmap on kernels where this is
  needed

Fix #976
Fix #3082
commit 5afa7ba7 (2019-01-04): core: support unmap of managed dataspace generally

This commit solves several issues:

* correct calculation of overlap region when detaching regions
  in managed dataspaces
* prevent unmap of Fiasco.OC's core log buffer
* calculate the core-local address of regions in managed dataspaces
  if possible at all and use it to unmap on kernels where this is
  needed

Fix #976
Fix #3082
closed 2019-01-07 by nfeske