genodelabs/genode #3086: sel4: test-libc always fails

skalk, 2018-12-12:
Due to the big amount of RAM used by the libc test to stress malloc, this test always leads to a resource problem on all Sel4 platforms we support. Either we skip that test right now on this platform, or make the test configurable, or fix the resource problem.
To silent the jitter of the test-infrastructure I would vote for the first option in the first-place, then open up an issue to make the test-libc configurable for platforms with less resources (rpi, foc_pbxa9,...), and in the long-term fix the resource problems of Sel4. Other opinions?

nfeske, 2018-12-12:
I agree with everything you just said.

skalk, 2018-12-12:
For hw/rpi the RAM resources are also not enough.

commit 7f90eade (2018-12-14): depot_autopilot: skip test-libc on rpi and sel4

Fix #3086
commit a322e8ab (2019-01-04): depot_autopilot: skip test-libc on rpi and sel4

Fix #3086
closed 2019-01-07 by nfeske